betaflight/betaflight #9559: Smoothness Set to 100 in Receiver Tab results on FULL THROTTLE on Arm!!

spatzengr, 2020-03-07:
See the video at: https://www.screencast.com/t/eGvrIyNdtVpz showing the issue.

[btfl_001.zip](https://github.com/betaflight/betaflight/files/4302463/btfl_001.zip)

etracer65, 2020-03-07:
If you can set it to 100 in the receiver tab then that's a bug in the Configurator. The setting is limited to 50.
```
rc_smoothing_auto_smoothness = 10
Allowed range: 0 - 50
```

etracer65, 2020-03-07:
Just tested the 10.7 configurator and it's limited to 50 so not sure how you set 100. Also can't view the video (flash player required and not installing it).

etracer65, 2020-03-07:
Ah I see, there's a bug in the Configurator that the "Save" happens before it constrains the range (if you type in a bit number).

etracer65, 2020-03-07:
So the problem is really in the Configurator and we need to open an issue there. But since we can't force people to upgrade to fix the problem in 10.6 we'll probably have to put in a validation check in the MSP to prevent it from occurring. Normally we don't add additional validation to MSP and it's the client's responsibility to validate, but there you go. Since this seems to be a bad result (although admittedly no something many people are going to fall into) we probably have to mitigate.

etracer65, 2020-03-07:
It looks like the range constraint checking is broken for all the fields in that section of the configurator. Fortunately the only other settings that aren't drop-downs are the manual cutoff frequencies and these allow 0-255 so it's not possible to get too bad a result.

etracer65, 2020-03-07:
Created https://github.com/betaflight/betaflight/pull/9560

etracer65, 2020-03-07:
Also created an issue for the Configurator: https://github.com/betaflight/betaflight-configurator/issues/1907

PR #9560 (2020-03-09): Add validation to MSP for the rc_smoothing_auto_factor setting
Fixes #9559 

This is a workaround for a validation bug in configurator 10.6 where the user can enter an invalid value, immediately click "Save" and that value will be stored before the value is range-checked. Added validation to the MSP message processing to constrain the range to prevent a dangerous result.