betaflight/betaflight #9566: FLYWOOF7DUAL UT error w/ DMA assignment

RipperDrone, 2020-03-08:
Seems UT file is mis-configured OR issue w/ current Nightly: To replicate, flash actual Nightly #2112, apply defaults, type defaults nosave in CLI --> throws following error:

# defaults nosave

# resetting to defaults
###ERROR: PARSE ERROR###
###ERROR: INVALID DMA OPTION FOR PIN C09: '0'###
###ERROR: INVALID DMA OPTION FOR PIN B08: '0'###

These pins are listed in resource overview as motor 5 and motor 6 pins. Looks like timer/DMA assignment is incorrect?

mikeller, 2020-03-08:
Heh, looks like Flywoo f'ed up their Unified Target configurations:

```
timer PB5 AF2
# pin PB5: TIM3 CH2 (AF2)
timer PB7 AF2
# pin PB7: TIM4 CH2 (AF2)
timer PC9 AF3
# pin PC9: TIM8 CH4 (AF3)
timer PB4 AF2
# pin PB4: TIM3 CH1 (AF2)
timer PC8 AF3
# pin PC8: TIM8 CH3 (AF3)
```

The error was in there for a long time, but we only now started to throw errors when applying custom defaults.

Can you try this (download the latest STM32F7X2.hex from https://ci.betaflight.tech/, 'Load Local File', select the .config, 'Load Local File', select the STM32F7X2.hex, the 'Flash'):

[FLWO-FLYWOOF7DUAL.zip](https://github.com/betaflight/betaflight/files/4303015/FLWO-FLYWOOF7DUAL.zip)

RipperDrone, 2020-03-08:
@mikeller tried what u suggested:

then connected FC over USB - load defaults? yes - CLI defaults nosave --> 
###ERROR: INVALID DMA OPTION FOR PIN B08: '0'###

So c09 seems ok now, but b08 assignment not so.

RipperDrone, 2020-03-08:
Listings (including your fix) which might help:

# dma
dma ADC 3 0
# ADC 3: DMA2 Stream 0 Channel 2
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 0 Channel 6
dma pin A03 1
# pin A03: DMA1 Stream 3 Channel 6

# timer
timer C06 AF3
# pin C06: TIM8 CH1 (AF3)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer A15 AF1
# pin A15: TIM2 CH1 (AF1)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
**timer B08 AF3
# pin B08: TIM10 CH1 (AF3)**
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer A02 AF2
# pin A02: TIM5 CH3 (AF2)
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)

RipperDrone, 2020-03-08:
just semantics, but one comment correction is still inconsistent:
[image]

mikeller, 2020-03-09:
@RipperDrone:

> just semantics, but one comment correction is still inconsistent, should be C09 instead of PC9:

Thanks, well spotted!

> ###ERROR: INVALID DMA OPTION FOR PIN B08: '0'###

Damn, this is more broken than I thougth. **Do they even test?**

I don't have the board, can you test again:

[FLWO-FLYWOOF7DUAL.zip](https://github.com/betaflight/betaflight/files/4306033/FLWO-FLYWOOF7DUAL.zip)

mikeller, 2020-03-09:
Also, not sure why GitHub is closing issues when a linked pull request is force-pushed. :thinking:

RipperDrone, 2020-03-09:
@mikeller .config verified - doesn'T throw any error anymore. Seems fixed. Cannot judge on timer / dma settings being correct, but quad did fly ok even before changes.

Closing this issue...

closed 2020-03-09 by RipperDrone

mikeller, 2020-03-10:
Thanks for testing @RipperDrone!