betaflight/betaflight #9568: Rate Profile Selection

Prozay033, 2020-03-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

I think i found a bug in 4.1.4 and do not know who to contact, but i enabled rate profile on AUX 8, CH12 in BF, and also did what i had to do to get rate profile selections to work. the only thing is it will not save my desired profiles. for me id like to use this feature to be able to tune one quad on 4s and on 6s, as i have both. i do notice a difference when switching batteries and trying to find the right tune, and for me this feature is gold! thanks

Prozay033, 2020-03-09:
if there is anything i need to provide let me know as this is my first time thanks

Asizon, 2020-03-09:
You need to provide all of requested

closed 2020-03-10 by no-response[bot]