betaflight/betaflight #9581: [CRSF] Use normal "RSSI value" to display "RSSI dBm"

0crap, 2020-03-13:
I've read PR #9550 and there is a comment that says "You should still be able to get the RSSI dBm in place of RSSI - if you scale it right it will display dBm * -1."

I'm not able to get this to work correctly.
What I did:

On the BF Configurator OSD tab I have switched on:
- Link quality
- RSSI dBm value
- RSSI value

On the CLI I have configured:
- crsf_use_rx_snr = OFF
- rssi_scale = 100
- rssi_invert = OFF

With this I see the following happen in the OSD:

RSSI dBm                        RSSI value
-80                                  37
-70                                  45
-60                                  52
-50                                  61
-40                                  68
-20                                  84

I don't think it's possible to make the RSSI value show negative numbers.
Should it be possible to configure this to get:

-80                                 80
-70                                 70
and so on?

0crap, 2020-03-14:
OK, I played a bit more with all rssi CLI settings I could find.
It really seems not possible to mimic RSSI dBm in the RSSI value.

I can live with that, but it should not be mentioned that this is possible.
Let me know if anyone else tested this and is able to get better results.
Otherwise I'll close this issue.

mikeller, 2020-03-14:
@0crap:

> RSSI dBm -> RSSI value
-80 -> 37
-70 -> 45
-60 -> 52
-50 -> 61
-40 -> 68
-20 -> 84

Considering that the [-130, 0] range of the CRSF RSSI dBm is scaled to [0, 100), this is what is to be expected:

100 - (-80 / -130 * 100) => ~37
100 - (-70 / -130 * 100) => ~45
[...]

Doing it this way will give you output for the full range of -130 to 0 inside the [0..99] range that is displayed for RSSI. This will also work for things like RSSI alerts in the OSD and over telemetry that all assume that 'smaller is worse' for RSSI.

If you want `-1 * RSSI dbm` to be displayed in OSD (obviously forfeiting useable values > 99), and not being able to use RSSI alerts, you will have to wait for #9586, and then you will be able to use the following settings:

```
set rssi_invert = ON
set rssi_scale = 130
set rssi_offset = -100
```

0crap, 2020-03-15:
Great! When a nightly build is available, I'll test and report back.

closed 2020-03-16 by mikeller

0crap, 2020-03-16:
@mikeller I tried it and run into the problem that the "RSSI value" OSD element is a blinking zero.
I guess the "RSSI value" OSD element is not allowed below zero?

mikeller, 2020-03-16:
@0crap: Wait, on second thought, you should not need the `set rssi_offset = -100`. Can you try just this:

```
set rssi_invert = ON
set rssi_scale = 130
```

0crap, 2020-03-16:
@mikeller Sure, and that setting looks promising!

RSSI dBm -> RSSI value
-70 -> 70
-80 -> 80
-90 -> 90
-100 -> 99
-110 -> 99
and so on...

The minus missing in front of the number is no problem.
But when the RF level is starting the get low and thus most interesting to know, you are stuck at 99.

mikeller, 2020-03-16:
As I said, you won't be able to go above 100 in the RSSI display. You could add a `offset = -30` to capture the -130 ... 30` range.

0crap, 2020-03-16:
OK, understood, that will give below:
-40 = 10
-50 = 20
-60 = 30
-100 = 70
-110 = 80
and so on...

Unfortunately that is not something my brain can deal with in flight.
But at least you can set an alarm on it and not think about it in flight.

mikeller, 2020-03-16:
Well, I don't think it is so much a matter of knowing the exact dBm value - it's more a matter of knowing what value is the minimum workable one, and that 'value goes down' is bad - I bet you'll be used to the new scaling in no time at all once you started using it. :wink:

0crap, 2020-03-16:
True, I guess you'll get used to it.
But for now I'll go with SNR and LQ.
(RSSI on top of it would be nice for a first few flights, if you're new to CRSF and want to learn how it all works.)
You can always toggle between SNR and RSSI dBm in the config to see what fits your needs best.

downfold, 2021-01-09:
Hello there and I hope somenone, please, be kind and help me with a problem that I encounter today. 
I tried to back-up my via CLI panel, i set diff all....And I encounter a problem whit my RSSI dbm values. 
I'm new in FPV and I don't handle verry well the "language" and I don't what is caused my problem.

The code from CLI: ###ERROR: dump: CORRUPTED CONFIG: osd_rssi_dbm_alarm = 21###   (first the value was 196...now I cannot jump more than 131). 

Thank you and I hope you can help me. 

I curently use betaflight 4.2
Other: Info Configurator: 10.7.0
Firmware: BTFL 4.2.3
Target: DIAT/FURYF4OSD(STM32F405

PR #9586 (2020-03-16): Made RSSI inversion apply to all RSSI sources.
Fixes #9581.

This will make it possible to configure RSSI to display the CRSF RSSI dBm * -1 value.