betaflight/betaflight #9593: [CRSF] osd_rssi_snr_alarm

0crap, 2020-03-15:
In BF 4.2 it is possible to display in the OSD "RSSI dBm value".

On the CLI the behaviour of this element can be changed by typing "set crsf_use_rx_snr = ON".
This element will now display the Crossfire SNR value. (Great!)

What seems to be missing is the possibility to set an alarm on this SNR value.
For example the CLI command "set osd_rssi_snr_alarm = <value>"
Allowed values: -6 - 60
Something like that.

Could that be implemented?
Or do I miss the obvious and is this already possible?

atomiclama, 2020-03-23:
@0crap I proposed a more independent solution https://github.com/betaflight/betaflight/pull/9540 But it was rejected as too much resource used. You could use something from that.

0crap, 2020-04-14:
@atomiclama Not really related. #9550 got merged but the missing part is you can't set an alarm for it.
It would be helpfull if that can be implemented.

atomiclama, 2020-04-15:
@0crap "not related" err there is a settable alarm in that implementation so yes I think it is related.  Like I said feel free to take any or nothing from it.

0crap, 2020-05-16:
Seems this is not needed, so I close it myself before this bot does it.

closed 2020-05-16 by 0crap