betaflight/betaflight #9598: Time out trying to flash - Version 4.0.6 on SPRACINGF3

daniel-snowsurf, 2020-03-15:
After upgrading F3 board SPRACING F3 clone to 4.0.6 successfully, I can no longer flash new firmware with configurator 10.6. Had to short boot pads. Several tests made with 4.0.6 version and 4.1.1 Performance Edition F3 (joelucid/betaflight). Going back to 3.5.7, I can flash normally with configurator.

Upgrade made with full chip erase. Problem exists without any further configuration. Configurator shows stm32 timeout error. After trying to flash new firmware and having the error,  it's not possible to connect with FC again. Board has to be reflashed by shorting the boot pads.

mikeller, 2020-03-15:
Seems to be a problem with your hardware - a lot of cloned boards are not accurate copies of the original and experience hard-to-diagnose problems like this.

daniel-snowsurf, 2020-03-15:
Thanks for your answer, I'll stay on 3.5.7 and flash with boot pads if needed. The FC is quite popular because it's from the Eachine Wizard X220.

mikeller, 2020-03-15:
@daniel-snowsurf: Eachine is well known for selling incorrectly cloned hardware and not supporting users of their hardware, so you are pretty much on your own to get this to work.

DiegoGPolo, 2020-04-16:
Hello i am new here, 
i tried to upgrade the firmware of a spracingf3 and i found the same issue, now i can load any version.
does someone knows how to fix it?

Thank you

Diego

closed 2020-05-23 by stale[bot]