betaflight/betaflight #9600: OSD Fonts failed to upload on DALRC F722 DUAL (BF 4.1.3, 4.1.4, 4.1.5)

TheMrJahn, 2020-03-16:
FC - DALRCF722DUAL. Configurator - 10.6.0 for MacOS.

After flashing to BF versions 4.1.3, 4.1.4, 4.1.5 and applying custom defaults it's impossible to upload OSD fonts (no matter what type of fonts). After flashing to BF version 4.1.2 (Legacy or DALR) you can upload fonts without any problem.

I have some more F7x2 FC (Matek, Talon) - this issue does not affect this boards.

Diffs:
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.2 Dec 22 2019 / 02:21:16 (89051e256) MSP API: 1.42
# manufacturer_id: DALR   board_name: DALRCF722DUAL   custom defaults: YES

# start the command batch
batch start

board_name DALRCF722DUAL
manufacturer_id DALR

profile 0

rateprofile 0

# end the command batch
batch end

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.5 Mar 16 2020 / 05:21:46 (d4e74e39c) MSP API: 1.42
# manufacturer_id: DALR   board_name: DALRCF722DUAL   custom defaults: YES

# start the command batch
batch start

board_name DALRCF722DUAL
manufacturer_id DALR

profile 0

rateprofile 0

# end the command batch
batch end

mikeller, 2020-03-16:
@TheMrJahn: Have you tried uploading the font with a battey plugged in? *Make sure to take the props off before trying.*

TheMrJahn, 2020-03-16:
@mikeller Just tried your solution and it doesn't help. Same issue with plugged battery.

TheMrJahn, 2020-03-16:
@mikeller Decided to test on all my F7 boards - TALONF7FUSION, MATEKF722, DALRCF722DUAL, KAKUTEF7, FPVM_BETAFLIGHTF7, IFF7_TWIN_G.
TALONF7FUSION, MATEKF722, KAKUTEF7, FPVM_BETAFLIGHTF7 - issue does not affect this boards.
DALRCF722DUAL, IFF7_TWIN_G - not able to upload OSD fonts with installed versions 4.1.3, 4.1.4, 4.1.5

etracer65, 2020-03-16:
Just tested with a DALRCF722DUAL with 4.1.5 and works perfectly. This board requires the battery to be connected as the OSD chip is not powered by USB. Test again and be sure to connect the battery first, then connect the USB. Trying to upload the font without the battery connected will result in the progress bar remaining stuck at 0% because the OSD chip is not powered.

TheMrJahn, 2020-03-16:
@etracer65 That's really strange - why can I upload fonts without battery using 4.1.2?

etracer65, 2020-03-16:
If you check you'll see that although the progress bar moves, the font really doesn't upload.

TheMrJahn, 2020-03-16:
@etracer65 Tnx for the explanation, need to test this more!

etracer65, 2020-03-16:
By the way, none of this is new information.

https://github.com/betaflight/betaflight/wiki/OSD-Font-upload-problem

TheMrJahn, 2020-03-16:
@etracer65 @mikeller Sorry, it's my fault, just checked again these 2 fcs by connecting the battery first - fonts uploaded successfully.

closed 2020-03-16 by TheMrJahn