betaflight/betaflight #9604: No Baro when using iFlight Succex F4 w/ Baro

calebdejournett, 2020-03-17:
So I have noticed this issue in which the built in Barometer is not recognized through betaflight on the iFlight SucceX F4 V2.1 TwinG w/Baro. I have attempted running every firmware 4.1.0 to 4.1.5 but nothing has worked. As well, I applied the set baro_hardware to every option hoping to get it to recognize something but nothing had happened. I had initially thought this might be a hardware issue, but upon installing an iNav 2.4.0 firmware onto the board, the baro started working. It had auto recognized it as a BMP280, trying to force this on a betaflight firmware yielded no results. I had attempted every troubleshoot I could find on the internet and through forum posts yet nothing worked. What is most confusing to me is the fact it works in iNav automatically but I cant even force it to recognize it in Betaflight.

```# version
# Betaflight / STM32F405 (S405) 4.1.5 Mar 16 2020 / 05:19:58 (d4e74e39c) MSP API: 1.42
# manufacturer_id: IFRC   board_name: IFF4_TWIN_G   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFF4_TWIN_G
manufacturer_id IFRC
mcu_id 00490032324d501520303256
signature 

# feature
feature SOFTSERIAL

# map
map TAER1234

# serial
serial 30 2 115200 57600 0 115200

# master
set baro_i2c_device = 0
set min_check = 1000
set max_check = 2000
set serialrx_provider = SPEK2048
set ibata_offset = 20
set gyro_2_align_yaw = 900

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0```

mikeller, 2020-03-17:
Looks like the I2C pin definitions for this board are different in iNav and Betaflight:

iNav:

```
#define I2C1_SCL                PB8
#define I2C1_SDA                PB9
```

Betaflight:

```
resource I2C_SCL 1 B10
resource I2C_SDA 1 B11
```

If the baro works in iNav then presumably the iNav one is right. Can you try this Unified Target configuration:

[IFRC-IFF4_TWIN_G.zip](https://github.com/betaflight/betaflight/files/4345461/IFRC-IFF4_TWIN_G.zip)

(Unzip, load with 'Load Firmware [local]', then select STM32F405 from the drop-down, 'Load Firmware [online]', then 'Flash Firmware', and 'Reset to Defaults' once flashed.)

calebdejournett, 2020-03-17:
Ive been working on this for days! Thanks for the fix, works wonderfully!

closed 2020-03-18 by mikeller

Linjieqiang, 2020-03-19:
So sorry @calebdejournett . It's my wrong. Thanks @mikeller .

JAL513, 2021-05-29:
Is it possible there is a similar issue with iflight SucceX-E Mini F7?  Baro doesn't register in betaflight. I've flashed the latest betaflight for the correct target firmware.