betaflight/betaflight #9605: Runaway with I-gain non-zero

whitenoise-fpv, 2020-03-18:
**Describe the bug**
A non-zero value for I-Gain in the PIDs makes my quad unflyable. I can hover my quad without Problems in acro/angle mode while having I-Gain zero. When setting I-Gain non-zero (even just a value of 1), then my quad goes to the moon in acro mode (in angle mode it does wobble so bad, that its unflyable) after barely touching the throttle. The motors are too hot to touch then after 10-15sec. When looking at the blackbox logs, you can see the i-term freaking out (it jumps instantly to 40% then down to -40% and up again) right when the throttle goes up (just 1-2 us!).

**To Reproduce**
My quad has the Mamba F722 mini stack incl. 4in1 ESCs, a 4inch frame, HQ 4x4.3x3, 6S 550mah lipos, R-XSR in Access mode via FPort. The problem occured at the dev version from March 15. Using identical settings in stable version 4.1.5 fixed the problem for me.
I am skeptical if that is reproducible (on another quad). 

**Expected behavior**
Yeah, using I-Gain should not shoot my quad to the moon (quite contrary it should stabilize my quad).

**Flight controller configuration**
```
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Mar 15 2020 / 01:34:03 (6ed4c0fbe) MSP API: 1.43
# manufacturer_id: DIAT   board_name: MAMBAF722   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MAMBAF722
manufacturer_id DIAT
mcu_id 004c00473237511734363730
signature 

# name: 4inch

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TREA1234

# serial
serial 0 0 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3

# aux
aux 0 0 3 1800 2100 0 0
aux 1 1 2 900 1800 0 0
aux 2 13 1 1700 2100 0 0
aux 3 28 2 1800 2100 0 0
aux 4 35 1 1300 1700 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880  591
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# master
set dyn_notch_q = 250
set dyn_notch_max_hz = 500
set acc_calibration = -110,53,-148,1
set max_check = 1950
set rc_interp = OFF
set rc_interp_int = 10
set rc_smoothing_input_hz = 40
set rc_smoothing_derivative_hz = 80
set rc_smoothing_input_type = PT1
set rc_smoothing_auto_smoothness = 20
set serialrx_provider = FPORT
set serialrx_inverted = ON
set rx_min_usec = 950
set rx_max_usec = 2050
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_idle_value = 300
set dshot_bidir = ON
set vbat_scale = 108
set ibata_scale = 107
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_cap_alarm = 400
set osd_tim2 = 1281
set osd_vbat_pos = 14721
set osd_rssi_pos = 2446
set osd_link_quality_pos = 35
set osd_rssi_dbm_pos = 98
set osd_tim_2_pos = 2455
set osd_remaining_time_estimate_pos = 248
set osd_g_force_pos = 8386
set osd_throttle_pos = 12441
set osd_vtx_channel_pos = 8595
set osd_current_pos = 8257
set osd_mah_drawn_pos = 2402
set osd_craft_name_pos = 10615
set osd_altitude_pos = 374
set osd_pid_roll_pos = 12617
set osd_pid_pitch_pos = 12649
set osd_pid_yaw_pos = 12681
set osd_pidrate_profile_pos = 6234
set osd_warnings_pos = 329
set osd_battery_usage_pos = 297
set osd_disarmed_pos = 2250
set osd_esc_tmp_pos = 364
set osd_esc_rpm_pos = 8376
set osd_rtc_date_time_pos = 362
set osd_flip_arrow_pos = 237
set osd_core_temp_pos = 396
set osd_log_status_pos = 8354
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set debug_mode = FFT
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5769
set vcd_video_system = NTSC
set displayport_max7456_wht = 3
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_harmonics = 2
set name = 4inch

profile 0

# profile 0
set dterm_lowpass_type = BIQUAD
set anti_gravity_gain = 1000
set iterm_relax = OFF
set throttle_boost = 0
set p_pitch = 28
set i_pitch = 1
set d_pitch = 13
set f_pitch = 29
set p_roll = 25
set i_roll = 1
set d_roll = 12
set f_roll = 27
set p_yaw = 18
set i_yaw = 1
set f_yaw = 45
set d_min_roll = 0
set d_min_pitch = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set yaw_rc_rate = 150
set yaw_expo = 1
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 30

rateprofile 1

rateprofile 2

# rateprofile 2
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 140
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 5
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 55

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

``` 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: CAMERA_CONTROL
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
*) the resource list is from the now installed 4.1.5 - including it just for the sake of completeness.

**Setup / Versions**
 - MAMBA F722 MINI Betaflight Flight Controller with 4in1 ESC 306 Blheli_32 from Banggood 
   ESCs at ver. 32.7 (48khz, timing 22)
 - T-Motor F1507 with HQ 4x4.3x3 v1s
 - FrSky R-XSR via FPort using newest Firmware RXSR-FPORT_ACCESS_191107 from Banggod
 - 6S, 550mah Lipo
 - no vtx, cam installed at the moment
 - not much wiring needed, just the rx to uart5, the motors to their ESCs, a 35v 470 cap and thats about it
 

**Additional context**
This is a new quad and those were the first hover tests. Blackbox logs exist, if needed. With BF 4.1.5 i managed to fly the maiden without problems (with the exact same settings posted above).

Hope this helps, to find a possible bug.

McGiverGim, 2020-03-18:
This was caused by this PR: https://github.com/betaflight/betaflight/pull/9569

Was fixed in master a little later, if you update it will be fixed.

whitenoise-fpv, 2020-03-18:
Very nice - like the improvements in 4.2, so i am happy to stay at the bleeding edge! 
Will update and check again.

Keep up the good work (y)

etracer65, 2020-04-14:
fixed in #9597

closed 2020-04-14 by etracer65