betaflight/betaflight #9611: MS5611 on H743 SPI bus doesn't work

MATEKSYS, 2020-03-19:
STM32H743VIT6,  MS5611
#define USE_SPI_DEVICE_3
#define SPI3_SCK_PIN            PB3
#define SPI3_MISO_PIN           PB4
#define SPI3_MOSI_PIN           PB5

#define USE_BARO
#define USE_BARO_MS5611
#define USE_BARO_SPI_MS5611
#define BARO_SPI_INSTANCE       SPI3
#define BARO_CS_PIN             PD4

[image]
[image]
[image]

jflyper, 2020-03-19:
@MATEKSYS 
I see that you have a board with a Rev.V silicon. Which board is that?

MATEKSYS, 2020-03-19:
> @MATEKSYS
> I see that you have a board with a Rev.V silicon. Which board is that?

Just a prototype, not released.

PR #9735 (2020-04-28): Fixed the MS5611 baro in SPI mode.
Fixes #9611.

The MS5611 gyro requires raw reads / writes when used over SPI. This pull request adds functions to do this to `src/main/drivers/bus.c`.

[image]

It also cleans up the ordering of functions in the MS5611 driver to get rid of forward declarations.
closed 2020-04-28 by mikeller