betaflight/betaflight #9618: Point Of Interest (POI)

scultek, 2020-03-21:
Hello,
Is it possible to add this flight mode please?

Thanks You,

mikeller, 2020-03-21:
@scultek: You'll have to provide more information than this. Please fill in the information that you were asked to provide when opening this:

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

scultek, 2020-03-21:
https://www.youtube.com/watch?v=IzjvDCA50DI
ThanksYou,

closed 2020-03-22 by mikeller