betaflight/betaflight #9627: MSP_MOTOR readout values not in sync with values former set by MSP_SET_MOTOR (Dshot/ProShot)

4712, 2020-03-22:
After setting values by MSP_SET_MOTOR value >Mincommand + 1 the values read back with MSP_MOTOR results in value -1.

This happens only, if Dhot or Proshot is set as protocol.
Here is what BLHleiSuite32 log shows after setting 1010 for all active motors:
```
MSP_SET_MOTOR: 1010 1010 1010 1010 0 0 0 0  OK
	send_msp:  OK
		$[24 4D 3C 10 D6 F2 03 F2 03 F2 03 F2 03 00 00 00]
		$[00 00 00 00 00 C6]
	recv_msp:  OK
		$[24 4D 3E 00 D6 D6]

MSP_MOTOR 1009 1009 1009 1009 0 0 0 0 OK
	send_msp:  OK
		$[24 4D 3C 00 68 68]
	recv_msp:  OK
		$[24 4D 3E 10 68 F1 03 F1 03 F1 03 F1 03 00 00 00]
		$[00 00 00 00 00 78]
```
**To Reproduce**
Use Betaflight Configurator
open Motor tab, set one motor to, let's say 1100
change to a different tab
return to Motor tab and recognize the value has changed to 1099 
Repeating tab changing reduces the motor value each time by 1. 

**Flight controller configuration**
```
# diff
# Betaflight / SPRACINGF3 (SRF3) 3.5.4 Dec 17 2018 / 12:33:18 (66b8ea143) MSP API: 1.40
set motor_pwm_protocol = DSHOT600
profile 0
rateprofile 0
```
```
# diff
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.5 Mar 16 2020 / 05:21:46 (d4e74e39c) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNINXT7   custom defaults: YES
# start the command batch
batch start
board_name OMNINXT7
manufacturer_id AIRB
# master
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 16
profile 0
rateprofile 0
```
The issue seems to have been going on for a long time, I don't know why I haven't come across it for so long.
Idea: rounding error in dshotConvertFromExternal or dshotConvertToExternal?

mikeller, 2020-03-22:
@4712: Yes, this is a known side-effect of how the rounding of PWM-ish input values into Dshot values and then back is done. There was an attempt to fix this a few years back, but the proposed fix to get to an idempotent mapping had other, more severe side effects, so it was decided that this was a minor annoyance without any relevant drawbacks, and it was left as it is.

4712, 2020-03-22:
@mikeller thanks for explanation. Makes sense. Will close the issue for now and search for workaround in the GUI.

PR #10223 (2020-10-06): Fix Dshot range
Retain motor PWM writes the same as PWM reads keeping DSHOT Command the same

Fixes: #9627
Fixes: https://github.com/betaflight/betaflight-configurator/pull/2183 

Beside switching between tabs it also fixes the following behavior when using sliders:

Dshot |   |   | Dshot 3D |  
-- | -- | -- | -- | --
Slider | Bar |   | Slider | Bar
1000 | 1000 |   |   |  
1001 | 1001 |   |   |  
1002 | 1001 |   | 1500 | 1500
1003 | 1002 |   | 1501 | 1501
1004 | 1003 |   | 1502 | 1501
1998 | 1997 |   | 1503 | 1502
1999 | 1998 |   | 1999 | 1998
2000 | 2000 |   | 2000 | 2000
closed 2020-03-22 by 4712