betaflight/betaflight #9630: URGENT! Battery Voltage display freezes when quad armed - Using  HD T-MOTOR F7 FC  WITH THE 55A 4IN1 32bit ESC. + DJI SETUP

btemtd, 2020-03-23:
This is an urgent issue as battery voltage is needed otherwise flying is hard to gauge how long left to fly.
Soon as I arm the quad, the battery freezes it doesn't read anymore until i unplug and then re-plug battery. Im quite lost  ( T-MOTOR F7 HD Flight controller WITH THE 55A 4IN1 32bit ESC. + DJI SETUP) I have 2 quads with this setup and both do the exact same thing both of them i flashed the latest firmware, i then re-flashed the old firmware back on and the problem still persists.

btemtd, 2020-03-23:
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.5 Mar 16 2020 / 05:21:46 (d4e74e39c) MSP API: 1.42
# manufacturer_id: TMTR   board_name: TMOTORF7   custom defaults: YES

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# serial
serial 1 33 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# aux
aux 0 0 0 1650 2100 0 0
aux 1 1 1 900 1250 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 35 2 1700 2100 0 0

# master
set gyro_lowpass2_hz = 188
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set small_angle = 90
set pid_process_denom = 1
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 2201
set osd_vbat_pos = 495
set osd_current_pos = 2071
set osd_mah_drawn_pos = 483
set osd_altitude_pos = 481
set osd_warnings_pos = 14601
set osd_avg_cell_voltage_pos = 2537
set osd_battery_usage_pos = 2058
set osd_stat_max_spd = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bb_no = OFF

profile 1

# profile 1
set dterm_lowpass2_hz = 113
set p_pitch = 45
set d_pitch = 36
set p_roll = 45
set d_roll = 33
set p_yaw = 90

rateprofile 0

# end the command batch
batch end

#

etracer65, 2020-03-23:
Contact DJI. They control retrieving the data and generating their OSD. It's not under Betaflight's control.

btemtd, 2020-03-23:
Really? How is it possible that Someone with Exact same setup just with the T-motor F4 everything is fine?... Like i said EVeryhting works I can see the voltage in the OSD Etc... But when I arm the quad the voltage stays but doesnt move anymore ( frozen number) When i unplug the battery and re-do It works again until i Arm the quad again. How can this be DJI If DJI Osd works with Betaflight? I am lost  dont even know what I am to ask them since it actually works but only when armed it freezes and it doesnt happen with the same exact setup only difference is the F4 FC

etracer65, 2020-03-23:
> But when I arm the quad the voltage stays but doesnt move anymore ( frozen number) When i unplug the battery and re-do It works again until i Arm the quad again.

Tell them exactly this. Like I said, DJI is polling (asking) the Betaflight firmware for information that they use to update their OSD. They are in control of it. This along with a bunch of other bugs they're dealing with at the moment in their current firmware. If you don't want to report this to DJI all I can suggest is that you try an older air unit and googles firmware.

btemtd, 2020-03-23:
Ahh ok thanks buddy... Is there any other work arounds like virtual mode? Just incase they may take time to fix this

btemtd, 2020-03-23:
ALso Is there a reason why my third quad which has a different FC and ESC , It uses the Lumeneir F7 Dual Gyro FC with the DJI air Unit aswell but the battery voltage on this does not freeze..  Only my 2 quads that have the dji T-motor stack.... does this still mean I need to contact or still DJI in this scenario now that you know this information.. or do you think its T-motor that needs to be told

etracer65, 2020-03-23:
I guess you could always use a timer and just land after a specific amount of time.

I don't see how there could be any relationship to the actual flight controller. More likely differences in firmware versions in the DJI air unit.

As a test try this. **WITH PROPS REMOVED!!** - connect a battery to your quad and then connect USB. Go into the configurator and on the motors tab enable the safety switch. Then go to the "Power & Battery" tab and make sure the battery voltage is displayed (also shown in the battery bar at the top). Arm your quad (again - **PROPS REMOVED!**) while still on the Power & Battery tab and confirm that the voltage readings are still being updated.

The configurator uses the same method to read the voltage from the flight controller as the DJI air unit does (by sending MSP message requests and processing the returned data from the flight controller). If the configurator continues to update it shows that there is no problem with this messaging.

btemtd, 2020-03-23:
Thanks for trying to help i really appreciate this . Ive been meddling for hours now its 4:30 am I just did what you said and everything is working perfectly , Voltage , amps, In every tab and above... Motors Tab, Power and battery Tab and Up the top ALL working PERFECT.. I hope this gives you an idea on a fix hehe because now im really confused. It cannot be the goggles because I have 3 Quads 2 with this strange problem and the other quad is also a DJI quad but has the lumeneir F7 FC and with the same goggles it works no issue at all

etracer65, 2020-03-23:
> I just did what you said and everything is working perfectly , Voltage , amps, In every tab and above... Motors Tab, Power and battery Tab and Up the top ALL working PERFECT.

Well this just confirms what I've been saying: It's not a Betaflight problem. Contact DJI support. The problem is not likely in the goggles, but in the DJI air unit firmware that communicates with Betaflight.

btemtd, 2020-03-23:
so its a bug which occurs only with DJI + f7 t-motor fc .. is that a thing that actually can happen, so DJI has to fix a certain bug which manifests only with certain FC?

btemtd, 2020-03-23:
I sent DJI an email....... The annoying thing is these 2 quads are my first ever builds the other Lumeneir F7 fc/Dji build i bought as an RTF thats why i keep thinking i have selected something wrong or maybe a certain switch in betaflight is conflicting with this specific setup.

etracer65, 2020-03-23:
We have no idea what bugs DJI has or how their air unit works. They don't share any information with us. So there's no point in continuing to ask what we think the problem is. Ask DJI.

Closing this as it's not a Betaflight problem.

closed 2020-03-23 by etracer65

btemtd, 2020-03-24:
Here is a 3rd quad i have with the DJI build i bought this RTF its called the Lumeneir QAV-R 2 ... has F7 Matek type stack... If anyone can see a difference in the CLI that may make one work and not the other please let me know

# 

# diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.1.1 Nov 15 2019 / 12:36:09 (1e5e3d369) MSP API: 1.42

# start the command batch
batch start

board_name MATEKF722SE

# resources
resource MOTOR 1 B01
resource MOTOR 4 B04

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# serial
serial 0 1 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1700 2100 0 0
aux 4 35 2 1300 1700 0 0

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 188
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 150
set dyn_lpf_gyro_max_hz = 375
set min_check = 1010
set max_check = 1990
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set beeper_dshot_beacon_tone = 3
set small_angle = 90
set pid_process_denom = 1
set osd_vbat_pos = 501
set osd_mah_drawn_pos = 2535
set osd_warnings_pos = 14602
set osd_avg_cell_voltage_pos = 2542

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 53
set dyn_lpf_dterm_max_hz = 128
set dterm_lowpass2_hz = 113
set anti_gravity_gain = 5500
set iterm_relax = RPY
set p_pitch = 45
set d_pitch = 36
set p_roll = 45
set d_roll = 33
set p_yaw = 90
set d_min_roll = 28
set d_min_pitch = 31
set d_min_boost_gain = 30

rateprofile 0

# rateprofile 0
set roll_rc_rate = 70
set pitch_rc_rate = 70
set yaw_rc_rate = 70
set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 15
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# end the command batch
batch end

#