betaflight/betaflight #9636: Horizon is wrong after turtle mode

vodka-bears, 2020-03-24:
**Describe the bug**
After using crash flip recovery in angle/horizon mode, arming leads to flipping back because of invalid horizon vector.
https://drive.google.com/open?id=1jBreezfBKGp7lU1pd-48sxcIPN1HLqwr

**To Reproduce**
Get a brushless micro quad/whoop, enable angle/horizon mode, crash upside down, use a turtle mode, arm.

**Expected behavior**
Horizon vector should match the actual gravity vector on arming.

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F411 (S411) 4.1.5 Mar 16 2020 / 05:20:24 (d4e74e39c) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF411RX   custom defaults: YES

# start the command batch
batch start

board_name MATEKF411RX
manufacturer_id MTKS

# name: VODKABEARS

# feature
feature TELEMETRY

# beacon
beacon RX_SET

# serial
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 0 1925 2100 0 0
aux 1 1 1 900 1400 0 0
aux 2 2 1 1400 1800 0 0
aux 3 35 2 1875 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 1
vtxtable powerlabels 25

# master
set gyro_sync_denom = 2
set mag_hardware = NONE
set baro_hardware = NONE
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330
set current_meter = NONE
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 2
set osd_vbat_pos = 385
set osd_rssi_pos = 2425
set osd_tim_2_pos = 2454
set osd_flymode_pos = 44
set osd_throttle_pos = 2401
set osd_crosshairs_pos = 2253
set osd_craft_name_pos = 2442
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2433
set osd_disarmed_pos = 2283
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set frsky_spi_tx_id = 37,153
set frsky_spi_offset = -47
set frsky_spi_bind_hop_data = 2,113,224,100,211,87,198,74,185,61,172,48,159,35,146,22,133,9,120,231,107,218,94,205,81,192,68,179,55,166,42,153,29,140,16,127,3,114,225,101,212,88,199,75,186,62,173,0,0,0
set frsky_x_rx_num = 4
set frsky_spi_a1_source = CONST
set gyro_1_align_yaw = 1800
set name = VODKABEARS

profile 0

# profile 0
set iterm_limit = 80
set angle_level_strength = 80

rateprofile 0

# end the command batch
batch end

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: RX_SPI_CC2500_TX_EN
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: RX_SPI_CC2500_LNA_EN
A14: RX_SPI_CC2500_ANT_SEL
A15: RX_SPI_CS
B00: ADC_BATT
B01: FREE
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: RX_SPI_EXTI
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
TIM3: FREE
TIM4:
    CH1: MOTOR 2
    CH2: MOTOR 3
    CH3: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
Stock Betafpv meteor65

**Additional context**
Cheers everyone

etracer65, 2020-03-24:
I use it all the time with my TinyHawkS (same target) and it works fine. Try waiting a second or two after flipping back over before rearming to give the attitude a chance to reset.

vodka-bears, 2020-03-24:
@etracer65 waiting is completely unacceptable when racing. It seems to me that Mahony filter isn't being reset when disarming after turtle mode.

etracer65, 2020-03-24:
The code looks correct to me and like I said, never have experienced this problem. Feel free to improve it.

vodka-bears, 2020-03-24:
@etracer65 okay, hope I can handle this.

vkolotov, 2020-03-25:
I can confirm, this is happening to me as well (TinyHawk2). I have to wait a sec/two before I can re-arm and take off, otherwise it it tries to flip back upside down.

etracer65, 2020-03-25:
That's normal and how it's supposed to work. After a crash (or even a hard turtle mode flip) the IMU attitude estimate can get out of sync because of the extreme g-forces (exceeding the sensor's range) and/or by too fast rotation (exceeding the gyro's range). When you disarm the logic waits for motion to stop (quad to come to a rest), then it speeds up the attitude convergence by putting more trust in the accelerometer. So there's short delay waiting for the quad to become motionless, then about 0.5 seconds to get the attitude to converge.

You just need to have a little more patience. Or alternately don't crash.

closed 2020-05-02 by stale[bot]