betaflight/betaflight #9639: remapping a uart to i2c does not work

ujjwalk450, 2020-03-25:
Currently I am using 2 different F7 FCs on my builds, the mamba f722 and clracing f7 dual. Both of the FCs have 6 uarts that according to datasheets are also having alternate functions of I2C. Like uart 1 on mamba is an i2c and uart 3 on clracing f7 is an i2c.

I would like an option to remap the unused uart to i2c for supporting baro and magnetometer for gps rescue purpose.

I tried to resource remap the uarts to i2c bus and tried attaching a bmp280 sensor and done all other requirements for calibration of barometer but it does not seem to work.

```# diff ALL

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.4 Feb  4 2020 / 17:27:35 (bc9715eec) MSP API: 1.42
# manufacturer_id: CLRA   board_name: CLRACINGF7   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF7
manufacturer_id CLRA
mcu_id 003000203237510731313935
signature 

# resources
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

# feature
feature -AIRMODE
feature ESC_SENSOR

# serial
serial 1 64 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200

# aux
aux 0 0 2 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 28 1 1700 2100 0 0
aux 4 35 3 1700 2100 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 7 16 25 40
vtxtable powerlabels 25 200 500 800

# master
set gyro_lowpass2_hz = 225
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 180
set dyn_lpf_gyro_max_hz = 450
set acc_calibration = -48,-1,34
set mag_hardware = NONE
set baro_i2c_device = 4
set baro_i2c_address = 118
set baro_hardware = BMP280
set rssi_channel = 12
set serialrx_provider = CRSF
set dshot_bidir = ON
set vbat_scale = 162
set deadband = 5
set yaw_deadband = 5
set osd_vbat_pos = 2282
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5658
set i2c2_pullup = ON
set gyro_rpm_notch_harmonics = 1

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 51
set pitch_srate = 51
set yaw_srate = 51

rateprofile 1

# rateprofile 1
set roll_srate = 51
set pitch_srate = 51
set yaw_srate = 51
set throttle_limit_type = SCALE
set throttle_limit_percent = 75

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save```

mikeller, 2020-03-25:
@ujjwalk450: I2C remapping is supported, as long as the pins that you are using support I2C (the ones you are using do). The problem is `set baro_i2c_device = 4` - you are trying to use I2C4 for the barometer.

ujjwalk450, 2020-03-25:
thanks for the support, everything works fine now, i will now close the issue

closed 2020-03-25 by ujjwalk450

JosueCastro11, 2022-10-26:
Hello @ujjwalk450 and @mikeller, currently i having the same problem as @ujjwalk450, but i have the UART 3 busy

So I was wondering if it is possible to do the remapping in the UART 4, which I have it is liberate

I am using the Mamba F722S 

https://cdn.shopify.com/s/files/1/0027/2708/4144/files/1_F722S_F50PRO.jpg?v=1577435261

Thank you