betaflight/betaflight #9646: OMNIBUSF4 blackbox problem

m4ximuel, 2020-03-26:
**Describe the bug**
about all the 4.2.0 what I see on betaflight configurator
cannot access the blackbox but log is working well.
the feature is working with 4.1.5

**To Reproduce**
just update to 4.2.0
And I can change type for log on OSD. but cannot on betaflight configurator. and also cannot extract log file when using 4.2.0

**Expected behavior**
Resource conflict

**Flight controller configuration**
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 Mar 26 2020 / 10:43:48 (5b374c6fe) MSP API: 1.43
# manufacturer_id: AIRB   board_name: OMNIBUSF4   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 001d00213436470437333638
signature 

# name: vx120-cf

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1600 2025 0 0
aux 1 1 0 1425 1575 0 0
aux 2 1 0 1875 2025 0 0
aux 3 2 1 1250 1300 0 0
aux 4 2 1 1475 1525 0 0
aux 5 2 1 1675 1725 0 0
aux 6 2 1 1925 1975 0 0
aux 7 27 1 1600 2025 0 0
aux 8 13 0 1250 1300 0 0
aux 9 13 0 1475 1525 0 0
aux 10 13 0 1675 1725 0 0
aux 11 13 0 1925 1975 0 0
aux 12 35 0 1200 1250 0 0
aux 13 35 0 1425 1475 0 0
aux 14 35 0 1625 1675 0 0
aux 15 35 0 1900 1950 0 0
aux 16 39 1 1425 1575 0 0
aux 17 39 1 1875 2025 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 100 200 350

# master
set gyro_lowpass2_hz = 450
set dyn_notch_width_percent = 0
set dyn_notch_q = 350
set dyn_notch_min_hz = 250
set dyn_notch_max_hz = 700
set dyn_lpf_gyro_min_hz = 360
set dyn_lpf_gyro_max_hz = 900
set acc_calibration = -111,46,-15,1
set rc_smoothing_derivative_type = PT1
set rc_smoothing_auto_smoothness = 7
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set ibata_scale = 680
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_vbat_pos = 353
set osd_rssi_pos = 353
set osd_link_quality_pos = 2401
set osd_tim_2_pos = 2423
set osd_throttle_pos = 391
set osd_current_pos = 392
set osd_craft_name_pos = 2453
set osd_debug_pos = 321
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2433
set osd_pit_ang_pos = 407
set osd_rol_ang_pos = 375
set osd_esc_rpm_pos = 12
set osd_core_temp_pos = 400
set osd_stick_overlay_left_pos = 23
set osd_stick_overlay_right_pos = 0
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5769
set vcd_video_system = NTSC
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 700
set name = vx120-cf

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 126
set dyn_lpf_dterm_max_hz = 306
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 270
set anti_gravity_gain = 10000
set pidsum_limit = 1000
set throttle_boost = 7
set throttle_boost_cutoff = 25
set p_pitch = 60
set i_pitch = 70
set d_pitch = 60
set f_pitch = 180
set p_roll = 65
set i_roll = 60
set d_roll = 65
set f_roll = 180
set p_yaw = 90
set f_yaw = 100
set d_min_roll = 45
set d_min_pitch = 45
set d_min_boost_gain = 30
set d_min_advance = 0
set ff_smooth_factor = 0
set idle_min_rpm = 26

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set tpa_rate = 80
set tpa_breakpoint = 1750

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: PREINIT 2
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: PREINIT 3
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

**Setup / Versions**
 - HGLRC F4 Zeus F4 FC AIO 15A

**Additional context**

mikeller, 2020-03-26:
@m4ximuel: Sorry, your reproduction is not understandable. Can you please provide step-by-step instructions on how to reproduce the problem.

m4ximuel, 2020-03-26:
> @m4ximuel: Sorry, your reproduction is not understandable. Can you please provide step-by-step instructions on how to reproduce the problem.

This issue only occurs in 4.2.
The blackbox is working, 
It is also possible to change the log type in flight by operating the OSD.
it is marked as having no built-in black box memory on betaflight configurator.
Therefore, the recorded log cannot be accessed in version 4.2.0.
However, it can be accessed by downgrading to version 4.1.5. Also, the log is recorded normally.

mikeller, 2020-03-26:
@m4ximuel: No, this is working fine. If you are using the unstable development version of Betaflight, you will also have to use the unstable development version of configurator:

[image]

m4ximuel, 2020-03-26:
> @m4ximuel: No, this is working fine. If you are using the unstable development version of Betaflight, you will also have to use the unstable development version of configurator:
> 
> [image]

[image]

of course, and I also did test about 'normal version'. same problem.

m4ximuel, 2020-03-26:
@mikeller Has the hardware boot order changed?
I found a way. Access is possible by plug battery before connecting the USB.
But It was worked without the way. I think it is not normal case.

mikeller, 2020-03-26:
@m4ximuel: Ok. I did
- install the version of the firmware that you are referencing;
- apply your configuration;
- run the version of configurator that you are using;
- change the language in configurator to Korean.

Result:

[image]

This looks very much like it's a hardware problem with your flash chip - do you have another board with onboard flash that you can test with?

m4ximuel, 2020-03-27:
@mikeller 

- After plug battery and connecting the USB, it works normally.
- In 4.1.5 and below, just plug in the USB and it works.
- All three of the other types of FCs with built-in memory work well with 4.2.

When booting from USB power, there seems to be a problem only in version 4.2.

mikeller, 2020-03-27:
@m4ximuel: Please read https://github.com/betaflight/betaflight/issues/9646#issuecomment-604744173

closed 2020-03-27 by mikeller