betaflight/betaflight #9648: Spektrum Sat not working under 4.1.5

fletch3d, 2020-03-27:
CLRFLIGHT F4 board with SPektrum DSMX Sat worked under prior version of 4.1.2.  Lost bind firmware update to 4.1.5.

Used Spektrum sat_bind = and does not go into bind mode

# name: Fletch_R5

# timer
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B04 NONE
timer B08 NONE

# dma
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B04 NONE
dma pin B08 NONE

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -BLACKBOX_ERASE

# map
map TAER1234

# serial
serial 3 1024 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 2 900 1300 0 0
aux 1 13 1 1700 2100 0 0
aux 2 35 3 1300 2100 0 0

# master
set gyro_sync_denom = 2
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_calibration = -17,75,-162
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SPEK2048
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_rate = 480
set beeper_frequency = 0
set deadband = 3
set yaw_deadband = 3
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1400
set osd_vbat_pos = 2434
set osd_tim_1_pos = 2455
set osd_flymode_pos = 398
set osd_current_pos = 2422
set osd_mah_drawn_pos = 2402
set osd_craft_name_pos = 2443
set osd_warnings_pos = 2410
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vcd_video_system = NTSC
set name = Fletch_R5

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: SERIAL_RX 4
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: PINIO 1
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: PREINIT 3
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: CAMERA_CONTROL
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
 CH1: DSHOT_BITBANG 2
 CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11:
 CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE  

**Setup / Versions**
 - Josh Bardwell - CLRFLIGHT F4

**Additional context**
Have swapped out working SAT's and still get the same result in not being able to set bind mode.

mikeller, 2020-03-27:
Sorry, there is no target CLRFLIGHT that is supported in Betaflight - you will have to contact the supplier of this target to get help.

fletch3d, 2020-03-27:
Sorry Mistyped.  CLRACINGF4.

mikeller, 2020-03-28:
@fletch3d: Can you please try if this fixes the problem for you:

[betaflight_4.1.6_STM32F405.zip](https://github.com/betaflight/betaflight/files/4396399/betaflight_4.1.6_STM32F405.zip)

Unzip the file, make sure you have 'CLRACINGF4' selected in the drop-down, then 'Load local file', and 'Flash'.

(`4.1-maintenance` with #9398 removed)

fletch3d, 2020-03-28:
Sorry Same result

mikeller, 2020-03-29:
Not much we can do then - the Bardwell F4 is a clone of the CLRACINGF4, and it seems to have some differences in the hardware design, one of which seems to be causing this problem.

rhinorap, 2020-04-18:
I also have this problem on my emax tinyhawk2 Genuine board is MATEKF411rx (MTKS). Works perfectly on 4.1.3 .....

mikeller, 2020-04-19:
@rhinorap: So the board you are using is a genuine MATEKF411RX?

rhinorap, 2020-04-19:
@mikeller:  It's the board fitted to my Tinyhawk 2. I Just flashed 4.1.4 - Speksat works on this version.

closed 2020-05-30 by stale[bot]