betaflight/betaflight #9650: Betaflight 4.0 LED_STRIP not working with SPRACINGF3EVO

goatzillax, 2020-03-29:
LED_STRIP is not working with 4.0 on my SPRACINGF3EVO target (eachine FRF3_evo_brushed sbus).

I flash the firmware with a full chip erase, disable transponder and enable led strip.

Then I go to the led strip configuration, number all 4 LEDs and set them to red.

It works with 3.5.7.

With 4.0 the LEDs change to yellow as soon as I reboot with LED_STRIP enabled, but after that they never change.  In fact it seems the configurator never saves the settings either.

https://www.youtube.com/watch?v=tA6oO6AZSCo

Let me know if you need to see my dumps, but it's a full chip erase and I only touch LED_STRIP and TRANSPONDER.

Tested against all 4.0.x except 4.0.1 which wouldn't even boot.

goatzillax, 2020-03-29:
```
diff
# Betaflight / SPRACINGF3EVO (SPEV) 4.0.6 Sep  1 2019 / 01:06:10 (2a64051a2) MSP API: 1.41 / FEATURE CUT LEVEL 8
batch start

feature -TRANSPONDER
feature LED_STRIP
profile 0

rateprofile 0

batch end

# resource
resource BEEPER 1 C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource PPM 1 A15
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource LED 1 B08
resource TRANSPONDER 1 A08
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 A15
resource ADC_BATT 1 A04
resource ADC_RSSI 1 B02
resource ADC_CURR 1 A05
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 C14
resource GYRO_EXTI 1 C13
resource GYRO_CS 1 B09

# resource show all
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: ADC_BATT
A05: FREE
A06: FREE
A07: FREE
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: PPM
B00: FREE
B01: FREE
B02: ADC_RSSI
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: LED 1
B09: GYRO_CS 1
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: GYRO_EXTI
C14: SDCARD_DETECT
C15: BEEPER
F00: FREE
F01: FREE
F04: FREE

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: LED_STRIP
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

```

mikeller, 2020-03-29:
@goatzillax: LED_STRIP can not work in your configuration since you did not configure at all. Please read the instructions on how to configure LED_STRIP.

closed 2020-03-29 by mikeller