betaflight/betaflight #9651: Crash High Score

buzkill420, 2020-03-29:
We all love flying. that's why we're here, but when you fly hard, you're going to crash hard. Wouldn't it be nice to have just a little something to look forward to when you do.? Maybe even some bragging rights!    

My idea is to incorporate a "Crash High Score" into betaflight. Something that would give you some number to "compare crashes" in a sense.  

Either when disarm switch is triggered or maybe use the accelerator and gyro to detect a crash and then "generate" a  high score based on....how many degrees of rotation you do on individual axis', multiplied by deceleration factor? Not sure what would be best.

Assuming battery doesn't come unplugged, showing it in post flight stats would be neat or even just recording it to black box would be a good feature.

etracer65, 2020-03-29:
Most crashes result in rotation rates (2000dps) or accelerations (either 8 or 16g depending on the sensor) that exceed the sensor range for the gyro and accelerometer respectively.

bizmar, 2020-03-29:
set osd_stat_max_g_force = ON is basically that already, but like etracer said it's going to max out.

closed 2020-05-06 by stale[bot]