betaflight/betaflight #9652: GPS elements disappear intermittent from OSD

0crap, 2020-03-29:
I could find a similar report with issue number #7645.

With the latest master, build 2153, I still see this happening.
Most of the time it works correctly, but sometimes not.
After I connect the lipo to the quad, the GPS related elements are not shown.

On the workbench you can see it works like expected.
[image]

From what I see with testing it's exactly the same behaviour as if the GPS is not configured (off), using the GPS switch on the Configuration tab.
[image]

I can't reproduce this reliable, so that makes fixing this bug hard.

The easy fix in my opinion is not to depend the GPS OSD elements if GPS is configured on the quad.
Can that be a solution?
Downside is that a user can put GPS elements on the OSD when GPS is not configured.
But who cares if anyone does that or not? Becomes boring quickly if you ask me.

Talking on BF Slack about this issue, it seems more users are affected by this bug.

mikeller, 2020-03-29:
@0crap: The GPS related elements are shown in OSD whenever a GPS sensor is detected (i.e. a GPS device is sending data to the flight controller).
Currently it is hardcoded that if there is no data received from the GPS device for 2.5 seconds, then the GPS is considered to be not present - this seems to be what is happening in your case.
I recommend you check your GPS wiring and power supply.

For things like GPS rescue to work reliably, dropouts of more than 2.5 seconds are pretty bad and will most likely cause problems, and the way the GPS OSD elements are shown is intended to make pilots aware of these problems.

The targets below are built with the GPS timeout increased to 5 seconds - please test them and see if this makes a difference - if yes this will confirm the theory above.

[betaflight_4.2.0_STM32F745_norevision.zip](https://github.com/betaflight/betaflight/files/4398765/betaflight_4.2.0_STM32F745_norevision.zip)
[betaflight_4.2.0_STM32F7X2_norevision.zip](https://github.com/betaflight/betaflight/files/4398766/betaflight_4.2.0_STM32F7X2_norevision.zip)
[betaflight_4.2.0_STM32F411_norevision.zip](https://github.com/betaflight/betaflight/files/4398767/betaflight_4.2.0_STM32F411_norevision.zip)
[betaflight_4.2.0_STM32F405_norevision.zip](https://github.com/betaflight/betaflight/files/4398768/betaflight_4.2.0_STM32F405_norevision.zip)

0crap, 2020-03-29:
OK, fair enough.
If there is a dropout of >2.5 Sec, would that be visible in flight?
If all works on first connect I never see a warning or lose sats that I can tell during flight.

Wiring seems fine and I measured the 5V rail with my DMM, all OK.
I'm using a BEITIAN BN-180 which in the ports tab gets configured from AUTO to 57600 bps.

Without changing anything I flashed the F7X2_norevision and pasted my diff all.
Seems to work fine on the bench.
I'll let you know the results!

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Mar 30 2020 / 00:47:08 (norevision) MSP API: 1.43
# manufacturer_id: MTKS   board_name: MATEKF722   custom defaults: YES

0crap, 2020-03-29:
Well, I got lucky, multiple attempts on the bench and bang, no OSD elements!
So you can leave that 2.5 Sec as it is. :-)

Looking closely at the GPS LED's, the blue LED is blinking slowly when OSD elements disappear.
The blue LED is blinking very fast when all is OK.
So in the ports tab I changed the bps from 57600 to 9600, that seems to fix it!
Because it's intermittent I'm not going to say it's for sure, but it looks promising.

Maybe the AUTO setting for GPS needs to be looked at?
My best guess is that 57600 which BF seems to pick is not optimal.

etracer65, 2020-03-29:
Sounds like you need to check/reconfigure your GPS unit's settings in the u-blox u-center application. A lot of the cheap GPS units are shipped with questionable settings (ones like the BN180, BN220, etc.). Running at a 9600 baud rate is not ideal as it will limit the max update rate to be at 1hz or slower. This tends to lead to position and heading problems. Also every u-blox GPS should be able to communicate at 57600 with no problems. But if it's a clone and not a real u-blox GPS chipset then all bets are off.

0crap, 2020-03-29:
I'm using a BEITIAN BN-180 and if it's a clone or not, I can't tell.
That goes for almost everything from China we use in this hobby.

I tested all possible bps settings in the BF Configurator and 9600 works always.
Other speeds sometimes work, sometimes no communication.
If communication is there it seems to work reliable, also on 57600.

From the limited docs I can find on the web, 9600 seems the default for most of the BEITIAN stuff we use. Others speeds are indeed supported.
I can't tell if this is a BF issue or a GPS unit issue.
All I know is others also run into this problem from time to time.

0crap, 2020-03-30:
I played a bit with ublox u-center and the BN-180 default is 9600 baud.
It is possible to change the default to e.g. 57600 baud, which remains for hot/warm starts.
Unfortunately it's not possible to save the new config on the BN-180. So after a cold start it's back at 9600 baud.
(Tip, get a BN-220, which is able to store a new config.)

The question is what is "Auto Baud" and "Auto Config" in BF trying to do?
It should be possible to talk to the Beitian and set it to the proper Baud speed.

It seems iNavFlight [https://github.com/iNavFlight/inav/issues/3908](url)  had a similar issue, might be interesting for BF devs to have a look.

jflyper, 2020-03-30:
Auto baud cycles through possible speeds trying to read valid data from the device. If not received in a short time window, it moves to the speeds.

It sounds like we should all avoid BN-180, which doesn't have config storage flash...

0crap, 2020-03-30:
Ahhh, so Auto Baud cycles through possible speeds.
So Auto Baud is not actively talking to the GPS to set specific values?

If I type in the CLI gpspassthrough and connect MiniGPS 1.4.2.exe to it, I can set every value I want.
It takes everything without a glitch.

[image]
[image]

Also setting the 'Fix Update Rate' to 5 is no problem.
The default is 1, so it should be set to 5 IMHO.

If the solution is to get a BN-220 I'm more than happy to do so.
But are we sure 'Auto Baud' and 'Auto Config' can't do slightly better here?

mikeller, 2020-03-30:
@jflyper:

> It sounds like we should all avoid BN-180, which doesn't have config storage flash...

That sounds reasonable - using an 'auto baud rate' implemented in this way for a device potentially supplying vital information sounds like trouble.

0crap, 2020-03-30:
It's not only about the BN-180 gents, all BEITIAN stuff we use defaults to 9600 Baud / 1Hz.
So all users should configure their GPS unit at the proper speed and update frequency and save config?
For example with the u-blox u-center tool?

If that is the way forward we should remove the Auto Baud slider from the BF Configurator.

0crap, 2020-03-30:
Another thing that I find weird is that when I power my FC using USB and go to the Ports tab, set Sensor input to GPS and select AUTO, hit Save and Reboot I find AUTO has changed to 57600.
Even when the GPS is not powered. (5V rail is not up on my FC when using USB.)
Why is that?

0crap, 2020-04-05:
@mikeller OK, I played a bit more with this.
I used a simple config and only changed the basics to test this better.
```
# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Apr  4 2020 / 11:49:36 (cb8ab7547) MSP API: 1.43
# manufacturer_id: MTKS   board_name: MATEKF722   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722
manufacturer_id MTKS
mcu_id 002700193237510d31313935
signature 

# feature
feature GPS

# serial
serial 0 2 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 400

# master
set acc_calibration = -7,18,24,1
set gps_provider = UBLOX
set osd_gps_speed_pos = 2317
set osd_gps_lon_pos = 2090
set osd_gps_lat_pos = 2181
set osd_gps_sats_pos = 2214
set osd_home_dir_pos = 2311
set osd_home_dist_pos = 2150
set osd_altitude_pos = 2256
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 1
set vtx_freq = 5732
set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
Exactly this config I used in latest master and 4.1.5
With the latest master you know I sometimes see no GPS elements at all and sometimes it works.
When it does not work the GPS shows a slow BLUE blinking LED.

Now on 4.1.5 I see similar behaviour.
Sometimes directly GPS elements in the OSD and sometimes not.
However in 4.1.5 GPS elements will show after 15 seconds just waiting!
Then they just pop-up in the OSD!

In 4.1.5 it happens at startup I see the slow BLUE blinking LED on the GPS, but after a short time it starts blinking faster and GPS elements are there.

All this is repeatable on the bench with the above config.
I have no clue why it sometime takes longer for the GPS to start communicating properly.

mikeller, 2020-04-05:
@0crap: Yes, GPS signal acquisition has become worse in 4.1 compared to older versions, so the GPS symbols would be shown more intermittently in 4.1. Have a look at #9194 for a discussion.

0crap, 2020-04-05:
I'm not sure that one is related, I did all my testing indoor on the workbench. No GPS coverage at all.
My tests is more about initial GPS elements to show up on the OSD. (Flashing 0 sats whatever.)
It seems BF 4.2 is not trying long enough to startup communicating with the GPS and just stops.
While 4.1.5 keeps trying (I think) and starts communicating after 15 seconds in my tests.

mikeller, 2020-04-05:
@0crap: Sorry, confused myself there - yes, elements showing or not is down to communication with the GPS unit working or not, not having a GPS fix. But some of the discussion in the thread was about the 'AUTO' speed setting for GPS, and this potentially not resulting in stable communications - maybe you want to check that, and disable it if it is enabled?

0crap, 2020-04-05:
[image]

From the diff all you can see I tested with Auto Baud to OFF. (Which is the default.)
In the Ports tab AUTO goes immediately to 57600 after Save and Reboot. (Without the GPS even powered.)

0crap, 2020-04-18:
OK, I ordered a new BEITIAN BN-220 to see if I had bad hardware.
Conclusion, nothing wrong with my hardware.

On Auto Baud exact same issue, sometimes you have communication (shows OSD GPS elements) and sometimes no communication (no OSD GPS elements).

BUT, now I'm able to program the settings of my GPS with ublox u-center AND save them.
I changed the port speed of the GPS from 9600 to 57600 and saved it in flash.
I turned off Auto Baud in the configurator. (UART port speed still set at 57600.)

No issues anymore!
Always GPS elements, I powered OFF/ON my quad many times.

So the way I see it is that the "Auto Baud" feature does not work reliably in BF 4.2

mikeller, 2020-04-19:
@0crap: Not surprised, the auto-baud 'feature' takes pretty much a 'hail mary' approach to make the communication work. We probably should just remove it as it will not result in a reliable connection to the GPS.

0crap, 2020-04-19:
@mikeller Glad you say it, exactly what I was thinking.
Please remove auto-baud it's total crap and give users the assumption it does something.
At a later stage look for a better implementation.

I can update the GPS wiki if you like.

mikeller, 2020-04-19:
@0crap: Not sure there is anything better. The approach to 'bootstrap' at a lower speed, and then synchronise the switch to a higher speed is flawed as it depends on all devices having exactly the same behaviour. The correct way to do this is to use a device that can persist non-default port speed settings, or else use the default setting.

0crap, 2020-04-19:
@mikeller Can inav example help with this? Not sure how they do it. Maybe better vs BF?

AirBreak69, 2020-04-19:
I like to share a note that may be off-topic but the described effect can overlay this issue:

Starting with build 2149 (introducing PR #9617, SBAS_NONE) I found intermittent failures of detecting the GPS module at all. OSD symbols are missing on startup, RTH not available.
I have a faked M8N which is a flash-less M8M. Besides: IMO many M8Ns modules do not have flash. The ublox receiver has an "auto detection" on its UART that makes the 'bootstrap' method working on this GPS module family.

Used configuration: static baudrate 57600, no Auto Baudrate option, Auto Config, SBAS none or off-area to avoid sat count drop.
Before that PR the detection was stable. Now the GPS module only is detected if
- The buffer battery is empty! On a bench (without chance of GPS reception) after some seconds of operation any consecutive power cycle failes the detection of the GPS module (OSD symbols disappear). One have to wait minutes or hours without power.
- Or the GPS receivers default baudrate of 9600 is used instead.

I found a workaround: increasing the GPS_TIMEOUT constant in gps.c from 2500 to somewhat 10000 solved the detection problem! (I didn't try lower values)
My conclusion: The startup time of the receiver seems to be prolonged with buffered memory available. Accompanied by the new configuration routines it is more likely to hit the timeout barrier.

aaaaaaudrey, 2020-04-24:
Is the state of this that currently, the only way to get stable GPS is to buy a flash-enabled GPS (bn220)?

mikeller, 2020-04-24:
@aaaaaaudrey: Either that, or use the GPS unit without flash at the 'native' port speed that it starts up with (normally 9600 baud).

aaaaaaudrey, 2020-04-24:
@mikeller I assume that this would drastically lower the gps update rate?

mikeller, 2020-04-24:
@aaaaaaudrey: 

> I assume that this would drastically lower the gps update rate?

Yes. Probably still good enough to use GPS for logging / speed display, but not for GPS rescue.

aaaaaaudrey, 2020-04-24:
@mikeller Aight. I just picked up a BN-220 to replace my TBS m8.

EggsBenedict, 2020-05-05:
This discussion has me a little confused ... I have a few M8N rebrands and they seem to run fine on 57600 (with auto-baud OFF).  Actually both run at 115200.  The ones that get flakey on 115200 are the Matek M8Q-5883's. I haven't checked with UBLOX center, but with auto-baud off but with the port set to 57600 or 115200 am I to understand that they're still running at their native rate despite the setting the ports tab?

FWIW, they seem to work very well - and GPS rescue functions well also.  The barometer equipped quads seem to manage better generally, but honestly with galilleo on and/or just simply high sat counts everything seems to do a pretty good job.

AirBreak69, 2020-05-05:
As far as I know the ublox M8N(M) receivers have an autodetect feature for the baudrate. So by simply talking to them at the desired speed, they switch the baudrate that they are using.
But the very first byte(s) may get corrupted on reception and may make the first command to fail. So it's likely that switching from native speed of 9600Bd to 57600Bd gives more time (and less corrupted command bytes) than trying 115200Bd.
This may be a reason for 57600Bd resulting in more stable connections. Besides: higher speeds are more likely to produce bit-errors due to bad wiring and noise.
So real M8Ns (with flash memory) preset to the 57600 or 115200Bd (manually saved to the flash settings) accompenied by the same fixed baudrate in BF should give the most stable detection of the receiver.
The used baudrate should be high enough the handle all the receivers data with 5Hz repetition rate assuming the highest number of satellites. Don't know if somebody was brave enough to calculate that. But 57600Bd seems to fit.

As mikeller stated correctly: With "auto-baud" BF tries to adapt to the GPS recievers actual sending baudrate by getting a meanfull message. IMO "auto-baud" only makes sense on receivers statically programmed to a non-standard or unknown baudrate.

EggsBenedict, 2020-05-05:
Ok. Thanks, @AirBreak69. That's along the lines of what I was hoping.

comsoft1908, 2020-05-06:
Had the same issue since moving to BF4.2 RC1 - I assume the problem is the auto baud detection.
I have 3 BN220 and all were set to different baud rates from the factory.

What worked for me: set the baud rate to 57600 via u-center, save the config with u-center. Go into betaflight configurator: disable auto-baud, enabled auto-config and set baudrate to 57600 in ports tab.

This way betaflight does not need to guess the baud rate, which fixes the problem.

sblaus, 2020-05-11:
Hi,Yesterday I update from 4.0.6(works all good) to 4.2 and GPS(beitian 220 ) coordinates disappeared on osd .

EggsBenedict, 2020-05-15:
> @0crap: The GPS related elements are shown in OSD whenever a GPS sensor is detected (i.e. a GPS device is sending data to the flight controller).
> Currently it is hardcoded that if there is no data [...]

FWIW - I tried this target on a PyroF4 board which intermittently has its GPS OSD elements disappear / the GPS isn't recognized (autobaud off, autoconfig on). I have two non-flash equipped GPSs (HGLRC M8N & RaceDayQuads M8N) that I have moved around a few different boards - they seem to work pretty flawlessly now on other boards I have, but on the Pyro board, **with or without the target above**, for some reason, **the GPS is often ignored**.  Moved it around onto different UARTS. Only have one Pyro board - but with the GPSs, have never had the OSD elements disappear with 3 other boards (Matek F722SE, Flywoo F722 Mini, and Revolt OSD Lite).

So odd. Not sure what to make of any of this. Seems more likely to happen once the board has been plugged in for a bit (and powering the GPS) and is power cycled rather than when cold.

(i'm sure just using a GPS with flash storage and configuring with ucenter is the best bet - something i'll keep in mind in the future).

mikeller, 2020-05-16:
@EggsBenedict: Can you provide a configuration `diff` for the board that proves to be problematic with the GPS?

0crap, 2020-05-16:
> (i'm sure just using a GPS with flash storage and configuring with ucenter is the best bet - something i'll keep in mind in the future).

100% fixed it for me, even used the same wiring harness, just swapped the GPS for one that can be set to a proper baud rate and set that as well in BF Configurator on the connected UART.

EggsBenedict, 2020-05-16:
@mikeller sure thing! diff below.
Looks like diff is from when i was testing with the targets provided above, hence the "NO CUSTOM DEFAULTS FOUND".

@0crap yeah - i believe that.  I just have 4 of these (two other Matek's without flash: M8Q's) and they work prettty much flawlessly on everything else that I have. All my mounts are made for this form factor, too. I'll get a 220 soon though to slowly test when I have free time. i drag my feet for a long time with anything that requires that I light up a virtual machine as I have a mac and ucenter is windows only.

`# 
# Building AutoComplete Cache ... Done!
# 
# diff
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F405 (S405) 4.2.0 Mar 30 2020 / 00:46:16 (norevision) MSP API: 1.43
# manufacturer_id: PYDR   board_name: PYRODRONEF4   custom defaults: NO

# start the command batch
batch start

board_name PYRODRONEF4
manufacturer_id PYDR

# resources
resource BEEPER 1 B05
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 C09
resource MOTOR 4 A08
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource INVERTER 1 C03
resource LED 1 B04
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 B09
resource CAMERA_CONTROL 1 B09
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource PINIO 1 A15
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04

# timer
timer B09 AF3
# pin B09: TIM11 CH1 (AF3)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B00 AF3
# pin B00: TIM8 CH2N (AF3)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)
[... 147 lines cut ...]
set d_roll = 90
set p_yaw = 55
set i_yaw = 80
set d_min_roll = 0
set d_min_pitch = 0
set motor_output_limit = 80
set ff_interpolate_sp = AVERAGED_4
set ff_spike_limit = 50
set ff_smooth_factor = 50

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 110
set roll_expo = 48
set pitch_expo = 48
set yaw_expo = 45
set roll_srate = 61
set pitch_srate = 61
set yaw_srate = 60

# end the command batch
batch end`

mikeller, 2020-05-16:
@EggsBenedict: The diff looks all good, not sure what is causing problems there - but I couldn't find any documentation on this board so I don't know if there are any switchable inverters or suchlike on this UART - have you tried a different UART on the same board?

EggsBenedict, 2020-05-16:
@mikeller yep! on this board, I tried (nothing made a difference).

While in the quad with the VTX/RX/Cam, etc.. attached:
1. Powering the GPS via 3.3v vs. 5v.
2. Powering first via USB and then battery.
3. Powering only with the battery.

Pulled from the quad with the bare board essentially only the GPS attached and using USB to power the board: 
1. All available TX/RX UARTS: 2, 6 & 4
2. All sorts of different configs: autobaud, no-autobaud, different fixed bauds.

This probably happens once every 6-8 plugins. The real reason I tried target above was because if i connected to BF quickly enough, I could catch the GPS icon turning on and the delay with which that icon would turn on, _seemed_ to be variable. But this is merely an observation ... I can't say the delay was actually variable at all given that the time it takes me to connect is uncontrolled. But that made me think it might be a time out issue.

I'm not toooooo worried about this on this setup, but if there's any testing that I can do to try and help. Please let me know.

mikeller, 2020-05-16:
@EggsBenedict: Thanks for testing and feedback - I was really clutching at straws, hoping to see a pattern that would allow narrowing down what causes these problems. But it does not look like there is one in your case - some boards work, others don't with the same GPS unit. :-(

closed 2020-06-26 by stale[bot]

macelius, 2021-05-09:
For me, it was a loose, but partially connected connector.