betaflight/betaflight #9664: Level_Race_Mode naming and adding it to Modes

ProjectMockingbird, 2020-03-31:
NAMING
NotFastEnuf (Travis Schrock) is the inventor of this type of flying and the name should be NFE_Racemode or Racemode.

REASONING:

1. **NFE Racemode** or **Racemode** is what it is commonly known on other flight control software, and we shouldn't confuse people (Quopa, Meow Mode anyone??)
2. It can also be used in Horizon Mode, so "Level" is confusing
3. If attribution is given to NFE in code, "Racemode" would acceptable

MODE
In this, it should probably be a mode and not just a Profile based setting.  There is enough diversity in usage on other platforms between Angle and Horizon users that being a mode is much more powerful.  A lot of people use it situationally (observing its use in Silverware users) so, Profile switching is less useful.

phobos-, 2020-04-07:
I'll work on renaming it.
@mikeller @etracer65 do you think it's good idea to rewrite it as a flight mode instead of profile setting?

mikeller, 2020-04-07:
@phobos-: The current name is hits the mark on being self explanatory and distinctive, which is how naming is done, so there is no reason to change it.
But we can definitely look into making this into a mode in 4.3.

qreacher, 2020-04-07:
@mikeller respectfully disagree that "level" is self explanatory as it can be used in Horizon mode as well as "Level" mode (called STAB in OSD, but everyone calls it Angle...much confusion and no definitive pattern across BF).  

if you consider "Level" to be the parent of Horizon, then _maybe_ this works, but_ I don't think the average user groups them as such and I have already had to answer the "will this work in Horizon mode" for early testers of 4.2 :)

But thanks for looking at it, and hopefully it makes it into Modes in 4.3.  I'll be promoting it a lot for Horizon updates in PMB so good timing on adding it to 4.2!

ProjectMockingbird, 2020-04-07:
Sorry, if you got a notification about this already, I accidentally logged in under my old old account :P

@mikeller respectfully disagree that "level" is self explanatory as it can be used in Horizon mode as well as "Level" mode (called STAB in OSD, but everyone calls it Angle...much confusion and no definitive pattern across BF).  

if you consider "Level" to be the parent of Horizon, then _maybe_ this works, but_ I don't think the average user groups them as such and I have already had to answer the "will this work in Horizon mode" for early testers of 4.2 :)

But thanks for looking at it, and hopefully it makes it into Modes in 4.3.  I'll be promoting it a lot for Horizon updates in PMB so good timing on adding it to 4.2!

mikeller, 2020-04-08:
@ProjectMockingbird:

> respectfully disagree that "level" is self explanatory as it can be used in Horizon mode as well as "Level" mode (called STAB in OSD, but everyone calls it Angle...much confusion and no definitive pattern across BF).

I think 'level' is not a bad match for the entirety of auto-levelling modes (i.e. horizon mode and angle mode).

And yes, 'STAB' in OSD is an anachronism stemming from the time when there were only acro and 'stabilised' mode (now angle mode), and should be fixed.

closed 2020-05-15 by stale[bot]

BISOPBISOP, 2022-06-21:
UP! :(