betaflight/betaflight #9667: Add numeric %capacity to OSD

Bhorski, 2020-04-01:
This would be a "nice to have" replacement for numeric "battery current mAr drawn" display in OSD.  The request is for a numeric display of the "Battery usage" OSD element that is currently a graphic (bar).

For those of us who fly by remaining battery capacity, and regularly fly the same battery configuration, % capacity remaining is a critical and accurate measure of when you are approaching battery damage.

Specific request: Allow an additional row in the OSD tab to select a new "Battery usage (numeric)" option that you can position in the OSD just like "Battery Voltage" or "Battery average cell voltage".  Format would be integer XXX% indicating remaining % capacity.

One alternative I've considered is just using the existing bar and not request this change. However the number is already computed and should be relatively simple to display numerically just as numeric voltages and currents can be displayed numerically in the OSD.

I've taken a stab at adding the feature to a screen shot of my default display.

[image]

Thanks for considering this.

Steve Gentry

mikeller, 2020-04-01:
There is a pull request already open for this: #9637

Bhorski, 2020-04-01:
Ok. Thanks!  I'll close this and look for a future release.

closed 2020-04-01 by Bhorski