betaflight/betaflight #9668: rpm_filter.c: motorFrequency var updated after frequency var

francelico, 2020-04-01:
Hi,

I was going through the rpm_filter.c file and noticed something odd, the motorFrequency var is updated at line 224 with its corresponding filtered rpm value:
https://github.com/betaflight/betaflight/blob/1b5b06197b3faad47b675a34201d95cc974bfaa4/src/main/flight/rpm_filter.c#L224

However, the frequency var that is used by the Notch filter method to be filtered out is updated at line 198.
https://github.com/betaflight/betaflight/blob/1b5b06197b3faad47b675a34201d95cc974bfaa4/src/main/flight/rpm_filter.c#L198-L199

The motorFrequency var is initialised at line 182, but with no values:
https://github.com/betaflight/betaflight/blob/1b5b06197b3faad47b675a34201d95cc974bfaa4/src/main/flight/rpm_filter.c#L182

which means that the first filter (in this case, gyro[x][motor0][harmonic0]) would be filtering out a random value, whatever is in the memory at motorFrequency[0] at the time. @joelucid @mikeller could you help me understand if I am missing something? Thanks.

Sam

mikeller, 2020-04-02:
@francelico: What you are missing is that motorFrequency is defined at unit scope (i.e. outside of a function), which means that it is persistent and initialised (defaulting to 0 in this case). So you are right that in the very first iteration of the loop the values of `motorFrequency` will be 0 - but then for every subsequent iteration this is not the case any more.

francelico, 2020-04-02:
Hi @mikeller , thanks for the fast answer, I understand much better now. I do have one more question then, what is the reason you are updating the motorFrequency variable afterwards? With the current implementation you are always filtering out the motor frequencies obtained during the previous pid loop, why this choice? 

The first thing that comes to my mind is that the gyro data has a slightly higher latency than the dshot telemetry data, either because of the physical response of the system (the motor causes the vibration measured by the gyro) or because of a higher signal latency. Therefore it would make sense to implement the logic this way to minimise the offset between the two.

mikeller, 2020-05-02:
@francelico: Sorry, got distracted. If you study the code, you will see that we are not updating `motorFrequency` every time `rpmFilterUpdate()` is run, but only every time all filter updates for a given motor are done - these filter updates are spread over multiple loop iterations to stay within the time budget for every loop iteration.

closed 2020-06-09 by stale[bot]