betaflight/betaflight #9669: With VTX Low Power Disarm enabled, armed Power Level max is 2

OGreaves, 2020-04-03:
DESCRIPTION:
With the vtx_low_power_disarm is set to "ON" or to "UNTIL_FIRST_ARM", the VTX power output only behaves as expected if set to Power Level 1 or Power Level 2.

TO REPRODUCE:
1. Unlock the VTX. Set up UART for SmartAudio. Determine VTX SmartAudio version. Set up the VTX table for that SmartAudio version, the corresponding Power Level format and the available VTX power levels (the VTX may require 3 or more Power Levels for this problem to become evident).
2. Set vtx_low_power_disarm = to "ON" or to "UNTIL FIRST ARM".
3. Set VTX Power Level to level 3 or higher (tested with a VTX that has 4 power levels)

EXPECTED BEHAVIOR:
When first powered on, after startup the VTX should transmit at Power Level 1. Once the motors are armed, the VTX should transmit at the power level selected in step 3 above.

ACTUAL BEHAVIOR:
When the motors are armed, if
VTX set to Power Level 1: Transmits at Power Level 1
VTX set to Power Level 2: Transmits at Power Level 2
VTX set to Power Level 3: Transmits at Power Level 2
VTX set to Power Level 4: Transmits at Power Level 2

BTFL 4.1.5 DIFF:
'''
diff

# version
# Betaflight / STM32F745 (S745) 4.1.5 Mar 16 2020 / 05:21:22 (d4e74e39c) MSP API: 1.42
# manufacturer_id: HBRO   board_name: KAKUTEF7MINI   custom defaults: YES

# start the command batch
batch start

board_name KAKUTEF7MINI
manufacturer_id HBRO

# name: Skyshell

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature LED_STRIP

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 16384 115200 57600 0 115200

# led
led 0 6,8::CT:1
led 1 9,8::CT:1

# aux
aux 0 0 0 1300 1700 0 0
aux 1 2 1 1300 1700 0 0
aux 2 13 0 1750 2100 0 0
aux 3 15 5 1300 2100 0 0
aux 4 26 6 1700 2100 0 0
aux 5 28 1 900 1700 0 0
aux 6 31 4 900 1100 0 0
aux 7 33 3 1600 2100 0 0
aux 8 35 2 1300 2100 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0
adjrange 1 0 6 1700 2100 29 6 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_sync_denom = 2
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set baro_hardware = AUTO
set rssi_channel = 13
set rc_smoothing_auto_smoothness = 50
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set align_board_yaw = 180
set bat_capacity = 650
set vbat_min_cell_voltage = 310
set vbat_warning_cell_voltage = 330
set vbat_scale = 109
set ibata_scale = 856
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_alt_alarm = 120
set osd_rssi_pos = 14403
set osd_link_quality_pos = 45
set osd_tim_2_pos = 14742
set osd_flymode_pos = 8290
set osd_vtx_channel_pos = 14369
set osd_current_pos = 14422
set osd_mah_drawn_pos = 14391
set osd_craft_name_pos = 14731
set osd_altitude_pos = 14721
set osd_debug_pos = 257
set osd_pidrate_profile_pos = 12419
set osd_warnings_pos = 14537
set osd_avg_cell_voltage_pos = 14455
set osd_log_status_pos = 8365
set osd_rate_profile_name_pos = 129
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_min_link_quality = ON
set osd_profile = 3
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5658
set gyro_1_align_yaw = 2700
set name = Skyshell

profile 0

# profile 0
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 65
set d_pitch = 40
set f_pitch = 0
set p_roll = 65
set i_roll = 90
set d_roll = 40
set f_roll = 0
set f_yaw = 0
set d_min_roll = 0
set d_min_pitch = 0

rateprofile 0

# end the command batch
batch end

# 
'''

RESOURCE SHOW ALL:
'''
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: LED 1
A03: FREE
A04: FLASH_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: LED_STRIP
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH1: LED_STRIP
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: LED_STRIP
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
'''

SETUP / VERSIONS:
Flight Controller: HolyBro Kakute F7 Mini ordered from PyroDrone.com on 3/15/2020
VTX: TBS Unify Pro HV (V1 running SmartAudio 2.0, unlocked)
-SmartAudio is connected to UART3

ADDITIONAL CONTEXT:
The tested system works correctly with vtx_low_power_diasarm set to "OFF", i.e. the VTX can be changed to any of the available power levels using SmartAudio through the Betaflight OSD.

OGreaves, 2020-04-03:
Seems similar to a problem reported in this comment: https://github.com/betaflight/betaflight...-368308468

MoppelMat, 2020-04-11:
> Seems similar to a problem reported in this comment: https://github.com/betaflight/betaflight...-368308468

the link is a 404 :(

Since when long does this problem exist?
I imagine that my power is drastically reduced since I started using this setting on 4.1.4...

OGreaves, 2020-04-11:
The link got truncated somehow, I'll try to post it again:
https://github.com/betaflight/betaflight/issues/5251#issuecomment-368308468

It was an indexing issue found and solved in BTFL 3.3. I don't know at what version the issue returned, I wasn't using higher power levels much this winter.

etracer65, 2020-04-14:
Tested using Betaflight 4.1.5 and a SA2.0 device. Cannot replicate, everything works as expected.

Perhaps you have some configuration issue with your VTX and it's locking out the higher power settings.

OGreaves, 2020-04-15:
Thanks for the feedback. Were you using an F7 based flight controller? When I turn off "VTX Low Power Disarm", I am able to switch to higher power levels using Smart Audio. I can't imagine what kind of configuration issue would cause that behavior.

etracer65, 2020-04-15:
Not that it has any bearing at all, but yes it was a F7 flight controller.

Note also that you are the only person reporting this problem so that further points towards your setup. Try a different vtx.

closed 2020-05-22 by stale[bot]