betaflight/betaflight #9676: GPS Return to home  issue on most recent 4.2 build

mactac, 2020-04-06:
**Describe the bug**
#1:  When using return to home, quad returns to home properly, but does so sideways instead of forward.  In other words, instead of pitching towards the home point, it rolls towards the home point.  It only started doing this after the upgrade to 4.2 from 4.14

**To Reproduce**
Enable RTH, use a GPS, hit RTH 

**Expected behavior**
Quad should point nose towards home point and pitch forward

**Flight controller configuration**
'''
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr  4 2020 / 11:48:32 (cb8ab7547) MSP API: 1.43
# manufacturer_id: KLEE   board_name: SYNERGYF4   custom defaults: YES

# start the command batch
batch start

board_name SYNERGYF4
manufacturer_id KLEE

# name: Mactac

# feature
feature -RX_PARALLEL_PWM
feature -LED_STRIP
feature -AIRMODE
feature RX_SERIAL
feature GPS
feature TELEMETRY

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 5 2 115200 57600 0 115200

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1300 1700 0 0
aux 3 46 3 900 1300 0 0
aux 4 28 0 1700 2100 0 0
aux 5 35 2 1700 2100 0 0

# master
set gyro_lowpass2_hz = 375
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 80
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = -97,-108,497,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 12
set rc_smoothing_derivative_type = BIQUAD
set rc_smoothing_auto_smoothness = 15
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_idle_value = 300
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 3
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 320
set vbat_scale = 101
set ibata_scale = 205
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_set_home_point_once = ON
set gps_rescue_angle = 35
set gps_rescue_initial_alt = 70
set gps_rescue_descent_dist = 50
set gps_rescue_throttle_max = 1700
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 2
set osd_rssi_alarm = 40
set osd_alt_alarm = 500
set osd_vbat_pos = 2081
set osd_rssi_pos = 2138
set osd_tim_2_pos = 14688
set osd_throttle_pos = 14649
set osd_current_pos = 2390
set osd_mah_drawn_pos = 2424
set osd_gps_speed_pos = 2368
set osd_gps_lon_pos = 2433
set osd_gps_lat_pos = 2449
set osd_gps_sats_pos = 2105
set osd_home_dir_pos = 2060
set osd_home_dist_pos = 2062
set osd_flight_dist_pos = 2073
set osd_altitude_pos = 2336
set osd_warnings_pos = 14602
set osd_avg_cell_voltage_pos = 2049
set osd_rcchannels_pos = 2000
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set osd_rcchannels = 4,5,6,7
set debug_mode = GYRO_SCALED
set vcd_video_system = NTSC
set gyro_1_align_pitch = 1800
set name = Mactac

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dterm_lowpass2_hz = 225
set anti_gravity_gain = 5000
set feedforward_transition = 40
set p_pitch = 64
set i_pitch = 60
set d_pitch = 53
set f_pitch = 60
set p_roll = 59
set i_roll = 60
set d_roll = 49
set f_roll = 60
set p_yaw = 47
set i_yaw = 70
set f_yaw = 60
set d_min_roll = 28
set d_min_pitch = 31

rateprofile 0

# rateprofile 0
set yaw_rc_rate = 150
set yaw_srate = 40

# end the command batch
batch end
'''

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` '''
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FLASH_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: PINIO 1
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: CAMERA_CONTROL
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
'''

**Setup / Versions**
 - Hobbywing RotorX Flight controller F4 G3
 - BN 180 GPS

**Additional context**
The problem only started happening with an upgrade to 4.2 from 4.14

mactac, 2020-04-06:
> @mactac: Are you actually able to reproduce this?

Yes.  I was out yesterday and it happened about 80% of the time.  
 
> Remember, this is GPS rescue, **not** 'return to home', so it is intended to be a last ditch effort to save a quad that was otherwise lost, and do so without attempting to do it gracefully.

Yes, I am aware of the intended purpose, I have been using it quite successfully for some time.

> So doing it while flying sideways is absolutely within the targeted performance for this.

That's interesting, because in my experience that doesn't seem to be true (at least not always).  Before, it would always turn around and point the "front" of the quad towards the home point and pitch forward to go home in a forward direction.

> My suspicion is that, for what you describe, your quad would have been going sideways at the time that GPS rescue activated. Since GPS rescue uses only information from the GPS, and the information on orientation that GPS can provide is solely based on movement, it will take whatever was the dominant direction of movement in the past to be the 'forward' orientation.

Well that's got me thinking - while I was testing, I wasn't moving around much, so it's possible that it could have been confused.  I will check that. That being said, I've never experienced a quad returning to home sideways before, and I've tested a lot.   It almost seems to be new behavior.    

The other thing I noticed was that the altitude was all over the place.   I realize that GPS is less than ideal for altitude, but this is the worst I've ever seen it, by far.  it was 100m off a few times, and sometimes I would be descending & the altitude would be rising just as fast as I was descending.   It was so bad I wondered if the GPS was malfunctioning.   Again, new with BF 4.2.

It's possible that my GPS broke right when I upgraded BF.... would seem like a bit of a coincidence though.

mikeller, 2020-04-06:
@mactac:

> Well that's got me thinking - while I was testing, I wasn't moving around much, so it's possible that it could have been confused. I will check that. That being said, I've never experienced a quad returning to home sideways before, and I've tested a lot. It almost seems to be new behavior.

I think the general assumption that GPS makes that 'a vehicle moves with its front pointing forward' is good - it holds true all of the time for most vehicles (cars, fixed wing aircraft), and will hold true for multirotor aircraft most of the time, especially when using FPV. But if you were 'strafing' for some time right before GPS rescue activation then it will get confused.

There were some changes made to the GPS configuration recently (discussion see #9194), and I am wondering if these changes resulted in a lowering of the threshold of how much a craft needs to move before the movement is used to derive orientation - this would explain why hovering is now more likely to result in bogus orientation than before.

mikeller, 2020-04-06:
@mactac:

> Interesting, I hadn't seen that. The problem I'm seeing is significant enough to warrant a deeper look if in fact that is what causes it.

As stated before, for the purpose of what GPS rescue is trying to do, 'flies towards the point of origin sideways' is still within the acceptable range - not ideal, but way better than losing the craft. So I don't think it _needs_ a fix.

> I wonder if this could also be the cause of my crazy altitude readings?

Potentially - maybe worth raising in #9194.

mactac, 2020-04-06:
> There were some changes made to the GPS configuration recently (discussion see #9194), and I am 

I also thought that this was interesting, because I'm having a similar problem... I go from 14 sats to zero, back up again, etc during flight in a wide open area.   Each time, I lose RTH capability... it's not easy on my heart when I'm 4km away :)

mactac, 2020-04-06:
> As stated before, for the purpose of what GPS rescue is trying to do, 'flies towards the point of origin sideways' is still within the acceptable range - not ideal, but way better than losing the craft. So I don't think it _needs_ a fix.

The challenge here is that the expectation is that RTH doesn't bring you all the way home (especially to land).   Th hope is that it brings you UP or BACK far enough for you to regain radio link/video/your composure/whatever  enough to regain control of the craft to pilot it home properly.

I'm not sure if you've ever had to deal with a craft coming back sideways, but it's not actually just sideways, it's sideways and "pitched" sideways, so you're looking out the "side" on a 35 degree (or whatever) tilted angle.   It's extremely hard to understand orientation in this situation & extremely difficult to understand even where you are.   It's also practically impossible to get a line of sight on where home is to estimate what you need to do to get home, plus you cannot see if you're about to hit something.

In my tests, it induced a bit of panic because it really is so difficult to figure out what's going on.  I would say that it's pretty critical to fix based on what I've experienced when this happens.

mikeller, 2020-04-07:
@mactac:

> In my tests, it induced a bit of panic because it really is so difficult to figure out what's going on. I would say that it's pretty critical to fix based on what I've experienced when this happens.

Again, as stated before, compared to the alternative outcome of you losing your craft, this outcome seems to be an improvement - apparently ou did not lose your craft. Getting this improvement is exactly what GPS rescue is designed to do, and is apparently doing. Yes, it can be improved by improving the likelihood of getting the orientation right even when the RC connection is lost while hovering, but I would not consider this 'critical'.

mactac, 2020-04-07:
> @mactac:
>
>
> Again, as stated before, compared to the alternative outcome of you losing your craft, this outcome seems to be an improvement - apparently ou did not lose your craft. Getting this [...]

Yes, of course it's better than nothing, but I am not arguing that it's not.   If you have ever actually experienced this, you would understand how critical it is.  I've experienced it, and it turned what is a pretty good lifeline (RTH) into something a LOT less effective.

etracer65, 2020-04-08:
Firstly I think your problem is simply a bad accelerometer calibration.
```
set acc_calibration = -97,-108,497,1
```
Is pretty unusual - particularly for the Z axis. Calibration is not just about "level", it also sets the compensation for any bias in the readings. So for example with your calibration on a flight controller that was setting level on the bench I see bias of a roll to the right (and slight pitch forward).

Secondly, GPS Rescue can't intentionally fly home any way other than forward. The flight control logic only has control of pitch and yaw. Roll is handled by the self-level in angle mode. Even if an incorrect heading was calculated due to initial hovering when activated, GPS Rescue will self correct. Initially it will yaw towards the incorrect home direction and then begin moving forward by pitching forward (all it can do). This motion will then allow the GPS to recalculate a good heading and GPS Rescue will yaw to correct the heading home.

So in your case I think it's only a problem with an incorrect accelerometer calibration making the quad roll when it thinks it's level. This then messes up the heading seen by the GPS unit as the craft is not moving straight forward and is instead heading off in an angle (due to the roll not being level).

flythebluesky, 2020-04-17:
I’m also seeing this on 4.1.5, KAKUTEF7. Yes, it brings the craft close to home. But it’s crabbing and angled up so there’s not much to see in the goggles, so you have to sense when it’s getting close - I can’t hear my 7” until it’s right on top of me. But this should not block 4.2 IMHO.

mactac, 2020-05-18:
> Firstly I think your problem is simply a bad accelerometer calibration.
>
> ```
> set acc_calibration = -97,-108,497,1
> ```
>
> Is pretty unusual - particularly for the Z axis. Calibration is not just about [...]

I just re-calibrated , and it comes up pretty much the same.  Z is the 3rd one I presume, 536?

It seems to fly fine, and when I put it in auto-level mode, it seems to be fine.

Since I'm not the only one having this problem, I wonder if there is some issue?   RTH is definitely coming home sideways.

closed 2020-06-26 by stale[bot]