betaflight/betaflight #9679: Dshot Beacon causes too much motor heat.

Ozzy33, 2020-04-08:
I been using MS since it came out, but recently tried dshot600. I enabled the dshot beacon and it is very nice being able to active via a switch (instead of waiting for the esc beacon timeout to end). But I notice the esc and motors get warm with in 10 seconds and hot within 30 seconds. They do not get this warm with MS (esc beacon feature), even if a lost quad/plane that is on for over a hour.

Looking at it today, the difference is the ON time the motor is being driven. With BLH_S, the beacon is on for ~0.3s and then off for ~2.5 seconds. roughly 10% ON time. Where BF dshot beacon is ON a much higher time percentage.

**Easy solution**, make one of the five beacon tones have lower ON time percentage (like 10~15%).

I could just blip the dshot beacon switch every couple seconds, but this won't work for rx_lost mode.

I read an article that shows a pulsing sound (or light) is easier to detect, specially when there is background noise (wind, traffic, etc).

Quad specs, BF 4.1.5, 4n1 35A tattoo BLH_S (v16.7FW) esc, 2207-2400, 5050x3, 4S.

Excellent work guys, appreciate it.

etracer65, 2020-04-08:
Betaflight doesn’t control this. Contact the BLHeli developers.

mikeller, 2020-04-08:
@Ozzy33: Good point - the interval time of the beacon tones will be a major factor in this. The way this is done is that Betaflight only sends the command to the ESC to sound one of 5 beacon tones, and it is the ESC that then generates the pattern that is defined on the ESC side. So it would really be up to the ESC firmware maintainers to add beacon tone patterns that have a lower on/off ratio.

Ozzy33, 2020-04-08:
Thanks guys, will do, you can close this now, fly on!

closed 2020-04-08 by Ozzy33