betaflight/betaflight #9680: Add support for LFR (lost frame rate) from FRSKY receivers in OSD

dlew381, 2020-04-08:
FrSky has added LFR (lost frame rate) % into the telemetry in recent receiver firmwares. This is similar to TBS Link quality. I can confirm this to be true in R9MM and R9Mini receivers. Please add support for this in the OSD.

mikeller, 2020-04-08:
@dlew381: Can you point to the documentation for how this has been added?

dlew381, 2020-04-08:
It's in the release notes in the attached readme file. I believe they've
added it to rxsr as well from what I've seen in rcgroups

Respectfully,
Dustin

*******************************Release Note******************************************

This Package is for R9M MINI/MM-OTA firmware update. 

Version and Files: 
R9MINI_OTA_LBT.frsk                                 LBT (certificate for CE )Firmware for R9M MINI/MM-OTA. 
R9MINI_OTA_FCC.frsk                                Non-LBT (certificate for FCC )Firmware for R9M MINI/MM-OTA. 
R9MINI_OTA_FLEX.frsk                               FLEX Firmware(No certificate) for R9M MINI/MM-OTA.

readme.txt                                                    Release note 

F.port / S.Port can be switched in the radio menu.

Firmware Version: v1.3.0

The release firmware resolved the issues below:
--------------------------------------------------------------------------------------------------------------------
1. Improved RF performance.
2. Added the telemetry for LFR (Lost Frame Rate).
-------------------------------------------------------------------------------------------------------------------
How to update R9M MINI/MM-OTA firmware :
By radio (SD card) :
1. Put the firmware under the folder [FIRMWARE] of sd card.
2. Connect receiver S.Port to the S.Port of radio.
3. Power on the radio and find the firmware,select it by press [ENT].
4. Select Flash xxx(receiver name), wait to end.
---------------------------------------------------------------------------------------------------------------------
More details please check FrSky website:
https://www.frsky-rc.com/product/r9-mm-ota/
https://www.frsky-rc.com/product/r9-mini-ota/

**********************All rights reserved to FrSky Electronic ., Ltd.*********************************

mikeller, 2020-04-08:
@dlew381: There is nothing in the release notes on how this can be accessed. Without technical details there is nothing that we can do to make this accessible in Betaflight.

dlew381, 2020-04-08:
Also, this is what it looks like in the telemetry screen on my Taranis.
Here they are calling it "FLR" instead of "LFR". LFR make more sense to me.

dlew381, 2020-04-08:
Ok I will reach out to the Frsky guy here in California and see if he can
get some documentation.

dlew381, 2020-04-09:
Another bit of information I found on FB. Apparently, the parameter is F010
and a user is putting it on an aux channel and inverting it as it is a
percentage. 100% represents no frames lost.

Still waiting on a response from FRSKY for documentation.

dlew381, 2020-04-09:
From FRSKY...

"Hi Dustin, 

That is an interesting idea. I will forward this to the other teams at FrSky for their review and input. Our engineers and software teams have worked with BetaFlight previously on implementing F. Port, so there may be something they can do in regards to this. If so, I would guess that they would work directly with BetaFlight or need BetaFlight to contact them about implementation. "

dlew381, 2020-04-10:
Alternatively, we could just give the user the option to assign an AUX channel to LQ like how we used to do with RSSI.

dlew381, 2020-04-15:
@mikeller FrSky was supposed to be in contact with BF Dev team one this. Presumably the same people that worked on f.port integration. Have they reached out? If not, I can follow up with the US rep.

mikeller, 2020-04-19:
@dlew381: No, we have not heard from FrSky yet.

jflyper, 2020-04-27:
> FrSky has added LFR (lost frame rate) % into the telemetry in recent receiver firmwares. >This is similar to TBS Link quality. I can confirm this to be true in R9MM and R9Mini receivers.

Frame losses in advanced digital transmission protocols may not be a correct index for radio link quality. This is especially true when the protocols have error correction capability in physical radio or datalink level, in which bit errors are masked at low error rate. In this case, frame error rate suddenly raises as error rate approaches to uncorrectable region.

Did you check to see if LFR changes linearly over the range?

dlew381, 2020-04-27:
I don't have the technical information to support this but my impression so
far is that it basically reports how many errors are the the last 100
frames and updates the valueb dynamically. I do know this function is being
added to most of the latest receivers as well as the newest ones being
released. I reached out to frsky a few weeks back to hopefully help
facilitate discussion to get this implemented. The US customer service rep
tells me that he has passed on the request to FrSky engineers but so far I
haven't heard anything.

dlew381, 2020-04-27:
@jflyper I do know of someone in the community that has done some testing on this. I will reach out and point him to this thread.

rinciong, 2020-04-27:
Hi All,  Dustin had asked that I chime in, but I have to preface this by saying my data was very subjective and anecdotal.

That being said, The LFR value does not seem to change with any sort of linearity with respect to distance, but the general feel provides enough feedback to help the user anticipate a failsafe, much better than gauging off RSSI.  Interesting enough, I've captured the relationship between LFR and the adaptive power settings of the R9 system, and they both seem to maintain a tight relationship.

I've noticed in areas of high noise, for instance, RSSI stayed fairly high, but LFR dropped significantly.  As a matter of fact, the relationship of LFR for a given location seemed to align with the LQ reading from my Crossfire setup, in that same location.  This itself had led me to believe that LFR behaved very similarly to the Crossfire LQ.

mikeller, 2020-04-27:
@rinciong: Thanks for the analysis. The last paragraph could mean that CRSF's 'LQ' and FrSky's 'LFR' are probably based on the same value reported by the RF chipset, considering they are both using the same chipset.

rinciong, 2020-04-27:
> @rinciong: Thanks for the analysis. The last paragraph could mean that CRSF's 'LQ' and FrSky's 'LFR' are probably based on the same value reported by the RF chipset, considering they are both using the same chipset.

That’s a very interesting point, and with all the opinions on the r9 vs Crossfire battle, I’ve never heard this concept before.

closed 2020-06-06 by stale[bot]

VitaliyNimych, 2020-07-13:
> FrSky has added LFR (lost frame rate) % into the telemetry in recent receiver firmwares. This is similar to TBS Link quality. I can confirm this to be true in R9MM and R9Mini receivers. Please add support for this in the OSD.

Hi!
In OpenTX add new data #define VALID_FRAME_RATE_ID       0xF010
https://github.com/opentx/opentx/blob/fea8d6952052ff70513878d6078332c69343d27a/radio/src/telemetry/frsky.h#L195
used to be called  "FRAME_LOSS_RATE_ID"

I think need add to 0xF010 https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/smartport.c#L153
and read this data

I can implement this if someone needs it, it’s not difficult to get this data and I see it in my telemetry TX -  VALID_FRAME_RATE_ID value "%"

[image]

VitaliyNimych, 2020-07-13:
See video  https://www.youtube.com/watch?v=d2Yf21Ct6PY&feature=youtu.be
I think you need to do it by analogy with RSSI, wrap the data in a feed and transfer back to FC in OSD