betaflight/betaflight #9683: Hobbywing Xrotor F4 G3: Mass Storage Device Mode is not working on macOS Catalina 10.15.4

bodja, 2020-04-10:
#### 1. Describe your problem;
Cannot activate mass storage device mode on mac os. **(on Windows works fine)**

#### 2. Include ways to reproduce the problem;

1. Format SD card to fat32.
2. Insert SD card into a flight controller.
3. Open "Betaflight-configurator" app.
4. Connect to the flight controller.
5. Navigate to the "Blackbox" tab.
6. Choose SD Card as logging device. Chose rate and mode.
7. "Save and reboot"
8. Connect the FC again
9. Go to the "Blackbox tab"
10. Click on the "Activate Mass Storage Device Mode" button.

Expected:
Disk is successfully mounted and accessible for usage.

Actual:
Disk partition is never mounted.

I've tried basic commands to mount.
```
$ diskutil list
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *31.9 GB    disk2
   1:                 DOS_FAT_32 BLACKBOX                31.9 GB    disk2s1

$ diskutil mount /dev/disk2s1
Volume on disk2s1 timed out waiting to mount

$ sudo mkdir /Volumes/BLACKBOX
$ sudo mount -w -t msdos /dev/disk2s1 /Volumes/BLACKBOX
mount_msdos: /dev/disk2s1: Resource busy
mount: /Volumes/BLACKBOX failed with 71
```

### 3. Software/hardware info
**Configurator**: 10.6.0
**Firmware**: BTFL 4.1.4
**Target**: AIRB/OMNIBUSF4SD(STM32F405)
**FC**: Hobbywing Xrotor F4 G3
**SD card**: SanDisk Extreme 32GB
macOS Catalina 10.15.4

Asizon, 2020-04-10:
I think that it is solved for 4.2 and 10.80
https://github.com/betaflight/betaflight/pull/9252

bodja, 2020-04-10:
@Asizon hi, I’ve tried v4.2 and had the same issue.

bodja, 2020-04-10:
@Asizon sry, haven't tried v4.2, it's not yet released. Is the release already planned?

Asizon, 2020-04-10:
Nah, this mentioned fix is aplied in 4.1.2, can you try 4.1.1?

bodja, 2020-04-10:
Have tried 4.1.1 and 4.1.2, the same issue

Asizon, 2020-04-10:
Ok, i think that the fix is aplied only for flash, maybe we have the same issue with sd card device @etracer65??

etracer65, 2020-04-10:
As requested when you opened the issue, provide the following information:

> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
>

bodja, 2020-04-10:
Initially this issue was created within the betaflight-configurator proj, that's why diff was not included.
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.4 Feb  4 2020 / 17:25:51 (bc9715eec) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# resources
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 A02
resource MOTOR 4 B01

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# aux
aux 0 0 3 1300 1700 0 0
aux 1 1 2 1300 1700 0 0
aux 2 13 1 1300 1700 0 0
aux 3 35 2 1700 2100 0 0

# adjrange
adjrange 0 0 0 900 1250 12 0 0 0
adjrange 1 0 0 1300 1700 12 0 0 0
adjrange 2 0 0 1750 2100 12 0 0 0

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 375
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = 3,19,-8
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_idle_value = 900
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2540
set osd_flymode_pos = 2276
set osd_current_pos = 2293
set osd_pidrate_profile_pos = 2244
set osd_avg_cell_voltage_pos = 2543
set osd_battery_usage_pos = 9
set osd_stat_battery = ON
set debug_mode = GYRO_SCALED
set gyro_1_align_yaw = 2700
set gyro_rpm_notch_harmonics = 1

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dterm_lowpass2_hz = 225
set p_pitch = 58
set i_pitch = 63
set d_pitch = 37
set f_pitch = 67
set p_roll = 53
set i_roll = 59
set d_roll = 34
set f_roll = 63
set p_yaw = 29
set i_yaw = 63
set f_yaw = 63

rateprofile 1

# rateprofile 1
set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 110
set roll_srate = 74
set pitch_srate = 65
set yaw_srate = 60

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 3
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 2
B01: MOTOR 4
B02: FREE
B03: PREINIT 2
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH3: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 2
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FRE
```

etracer65, 2020-04-10:
I just tested with a real OmnibusF4v3 (uses the OMNIBUSF4SD target) instead of a clone like the Hobbywing board with Betaflight 4.1.5 along with MacOS 10.15.4 and it worked normally. The sd card mounted on the desktop as expected. It's normal for it to take 10-15 seconds (or more) to mount depending on the size and contents of the card.

Have you confirmed that you can mount the sd card directly (removed from the flight controller?). Lastly try reformatting the sd card. Even though some cards ship pre-formatted there are cases where it's strange and won't work until reformatted.

bodja, 2020-04-10:
Reformatting has helped. Though the very first time I've formatted it in the same way. 

That is very strange. SD card work when I plug it in directly (removed from the flight controller) but `msc` does not work on mac and works on windows machine.

Looks like issue can be rejected.
If this happens again with this FC is it an FC problem or betaflight bug (so that I do not bother you when it's not a betaflight issue)

closed 2020-04-10 by etracer65

mikeller, 2020-04-11:
@bodja: What you describe is a common problem, and I've seen this happen in other devices using SD cards, like cameras: SD card works, then stops working. Reformat it, it works again. Working theory is that if there is an unstable connection between the SD card and the device (dust / dirt in slot, vibration, ...), a write operation is failing midway through, leaving the file system in a corrupted state, making it impossible for the operating system to read. And reformatting the card then fixes the problem.

bodja, 2020-04-11:
@mikeller makes sense, thanks for the clarification.