betaflight/betaflight #9685: HGLRCF405 invalid resource mapping after firmware upgrade

dm-mkl, 2020-04-10:
**Describe the bug**
After update of hglrc f4 flame FC to the new firmware one motor stops working.
RX and motor 4 don't work after upgrade
Looks like there is issue with default resource mapping.

**To Reproduce**
- all 4 motors work
- upgrade FC to the new version
Target: HGLRCF405 
Manufacturer: HGLR 
Version: 4.1.5 
Binary: betaflight_4.1.5_STM32F405.hex 
Date: 16-03-2020 07:50 
- apply custom defaults
- motor 4 stops working
- remap motor 4 from B05 to A02 pin
- motor 4 starts working

**Expected behavior**
- all motors should work after upgrade

**Flight controller configuration**
There isn't resources configuration before the upgrade.
After upgrade resources look like
```
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 B05
resource PPM 1 B08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 A08
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource CAMERA_CONTROL 1 B07
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource BARO_CS 1 B03
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource GYRO_CS 2 C14
```

**Setup / Versions**
 - hglrc f4 flame. ;
 - RX: frsky r-xsr[configured to UART1];
 - components wiring is attached
[fpv_drone_connection_diagram.pdf](https://github.com/betaflight/betaflight/files/4463235/fpv_drone_connection_diagram.pdf)

**Additional context**
Unfortunately I didn't find any information according to FC default resource mapping.
If there is information please provide.

Asizon, 2020-04-10:
Seems that you are using incorrect target, you need HGLR-HGLRCF405AS

dm-mkl, 2020-04-10:
Thank you for the explanation. You're right. 
HGLR-HGLRCF405AS works for me

closed 2020-04-10 by dm-mkl

Asizon, 2020-04-10:
No, current HGLRCF4 target is wrong, this resource need to be fixed, @HGLRC-D