betaflight/betaflight #9687: Can’t choose target devices nor the firmware in firmware flasher

ongeschikt88, 2020-04-11:
On MacOS  I can’t flash firmware for the flight controller. In the tab firmware flasher there are no targets in the dropdown box. 

It happens every time when I try to flash firmware. I use the latest betaflight configurator. Even reinstalled betaflight but still doesn’t work.

What I expected to happen was the same as you can find in al videos you can find online

I have a mamba f722 flight controller

mikeller, 2020-04-11:
@ongeschikt88: What is displayed in the log panel after you have clicked on 'Update Firmware'?

ongeschikt88, 2020-04-11:
I can’t even press that button. When I want to select a target the Dropbox keeps saying loading. But even after 6 hours it shows nothing

Nico

mikeller, 2020-04-11:
[image]