betaflight/betaflight #9704: Thrust Linear loses curvature at low values.

ctzsnooze, 2020-04-14:
Thrust Linear is intended to compensate for non-linear thrust, ie, less thrust at low throttle than at higher throttle.

It does this by providing a curved compensation factor, which boosts motor differential at low throttle, and attenuates at high throttle.  

Unfortunately the maths seems to be such that high amounts of Thrust Linear have a strong curvature, whereas small amounts are almost straight lines.  This means that small amounts of Thrust Linear cannot actually do what they are supposed to do.

We can see in this diagram that a large amount of TL causes strong PID gain at zero throttle, is neutral around 32% throttle, and attenuates above that.  There is marked curvature.  

[image]

When set to 10, the effect is much less, which is to be expected, but the curvature is almost entirely lost.  That's not good.

[image]

This is an example of motor thrust vs drive, showing the nature of the thrust curve for which Thrust Linear should be an inverse of....

[image]

The inflexion point appears to be around 30% throttle, above which thrust is fairly linear.

The amount of the correction required is small, maybe only 20% boost would be enough.  A TL value of 10-15 would be sufficient for a variation in responsiveness of that magnitude, but there is not enough curvature.

I'm flagging this because we probably should try to get more consistent curvature somehow.

KarateBrot, 2020-04-25:
Did you consider using a scale that weighs small values more? 
I was thinking about multiplying the thrust_linear value with a gamma curve to boost sensitivity for small values.

ctzsnooze, 2020-05-27:
Resolved with #9740

closed 2020-05-27 by ctzsnooze