betaflight/betaflight #9708: No command sent to ESC after entering dshotprog command into CLI. 4.1.0 Onward

conuthead, 2020-04-16:
Using furyf4 target. Entering a dshotprog command into CLI does not result in the command being sent to the ESC. ie: dshotprog 255 2. Dshot beacon works normally on switch.  Versions prior to 4.1.0 have dshotprog output. 

To reproduce, flash betaflight to f4 fury board ( maybe others ). Open CLI and enter any dshotprog command. All default betaflight settings. Tested on 3.5.7 good, 4.0.0 good, 4.0.6 good.
4.1.0 and 4.1.5 have no command sent .

etracer65, 2020-04-16:
Can you provide more info? For example, when testing is bidirectional DSHOT enabled or disabled. If enabled, are you using bitbang or timer-based bidirectional DSHOT? If using bitbang, can you try turning that off and use timer-based and test?

conuthead, 2020-04-16:
I am not using bi-directional dshot. No settings changed from default except changing dshot to 600, or 300 and pid time to either 4 or 8k.
The commands get send out normally when not sending them with dshotprog.  So flip over after crash and beacon are working fine but if you go to cli and try to send the command the FC does not send it in latest versions.

conuthead, 2020-04-16:
If it helps at all, I just tried proshot and its the same thing. Command is sent with versions < 4.1.0 only.

etracer65, 2020-04-17:
@conuthead I found the problem. Can you please test the attached version?

[betaflight_4.1.6_STM32F405.hex.zip](https://github.com/betaflight/betaflight/files/4490218/betaflight_4.1.6_STM32F405.hex.zip)

To flash the unified target first select your board type from the drop-down at the top (this loads the config for the board), then click the "Load Firmware [local]" button at the bottom and select this test version (be sure to unzip it first). Then flash your board. If you already have a 4.1.x version then you don't need to do a full chip erase and your settings will be preserved. If you have an older version already on the board then you should do a full-chip erase.

When testing initially PLEASE REMOVE PROPS! The fix was simple, but anytime we make changes to motor protocols there's always the risk of unexpected behavior.

In addition to testing `dshotprog`, also test non-CLI DSHOT commands like the beacon, turtle mode motor reversing, etc.

conuthead, 2020-04-17:
That did it ! @etracer65, dshotprog commands are being sent. I can not test turtle and motor reversing without CLI since I don't have transmitter hooked up but I can verify that the beeper is being sent by activating the switch range and also by sending `dshotprog`  commands.

etracer65, 2020-04-17:
@conuthead Thanks for testing. The fix is in #9712

PR #9712 (2020-04-20): Fix blocking DSHOT commands
Fixes #9708 

Motor and DSHOT refactoring broke blocking DSHOT commands as the check for commands being enabled was not taking into account that "blocking" type commands need to operate only when the motors are not enabled.

Fixes the CLI `dshotprog` command.
closed 2020-04-20 by mikeller