betaflight/betaflight #9710: Updated Succex F4 Flight Stack to 4.1, now only two ESC's work, BL_Heli also only reads ESC no3 and no4.

nikstojkov, 2020-04-16:
**Describe the bug**
Recently updated iFlight Turbobee 120 to betaflight 4.1, and for some reason two ESC's wont register at all in BLheli_S

I read the thread with an almost identical problem, however I am using the stated IFLIGHT_F411_PRO, but still no joy. Another worry is that i cant seem to find any other firmware to downgrade to to check (completely my fault for not taking note)

Everything else is functioning fine, checked soldered pins, used fully charges 2s Lipo's

Diff below

**To Reproduce**
Flash newest firmware )4.1.)

**Expected behavior**
All motors should spin

**Flight controller configuration**

 ``` 
# version
# Betaflight / STM32F411 (S411) 4.1.5 Mar 16 2020 / 05:20:24 (d4e74e39c) MSP API: 1.42
# manufacturer_id: IFRC   board_name: IFLIGHT_F411_PRO   custom defaults: YES

# start the command batch
batch start

board_name IFLIGHT_F411_PRO
manufacturer_id IFRC

# name: Turbo

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# master
set align_mag = CW270
set mag_align_yaw = 2700
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT300
set small_angle = 180
set pid_process_denom = 16
set gyro_1_sensor_align = CW270
set gyro_1_align_yaw = 2700
set name = Turbo

profile 0

rateprofile 0

# end the command batch
batch end
 ```
 
 ``` 
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: MOTOR 2
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH1: MOTOR 1
    CH3: MOTOR 2
TIM3: FREE
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
 ``` 

Not sure if im missing something glaringly obvious, or it just let the magic smoke out without me seeing.

N

etracer65, 2020-04-17:
Suggest you contact the manufacturer and ask them which target to use. If it's an "IFLIGHT" target and still doesn't work then they should review and confirm it's appropriate for the hardware they designed.

nikstojkov, 2020-04-20:
No response thus far from iFlight, however, I have just managed to semi repeat my issue with another iFlight board, but this time the problem came when flashing ESC's in BLHeli...

The newly bricked stack had 4.1 already installed on the FC, so I have a feeling that BF isn't to blame in this scenario...

dga711, 2020-05-08:
The SucceX-E F4 I have in my Nazgul wants BETAFLIGHTF4 as target

mikeller, 2020-06-01:
@nikstojkov: Tested with 4.1.5, with your configuration on an iFlight SucceX Micro F4 V1.5: All motor outputs work as expected:

[image]

This seems to be a problem with your hardware, either the flight controller or the ESC or the connector. Have a look to see if there is dirt / water in the tiny connector between flight controller and ESC, and make sure it is properly seated.

dukesdc5, 2020-07-07:
I fixed the exact same issue with my turbobee 136rs bnf. The only way I could get my FC to act right was finding the original dump and applying it..the problem is I cant remember if I had a backup or I refreshed to an older firmware. But the key was the dump. I almost threw the board out I spent hours trying to fix it.

closed 2020-08-16 by stale[bot]