betaflight/betaflight #9711: REVO NANO (OPEN) Target using wrong binary?

WillD7545, 2020-04-16:
**Describe the bug**
Flashing stalls halfway through flashing.

**To Reproduce**
Flash using (OPEN) Target

**Expected behavior**
A functioning flight controller

**Setup / Versions**
 - Genuine OpenPilot Revolution Nano bought from the OpenPilot website.
 - Loose flight controller

**Additional context**
I tried flashing before noticing the description said the REVONANO (OPEN) target is using a F405 binary instead of a F411 one. Flight controller seems to be softbricked maybe. Betaflight configurator can't detect it anymore.

mikeller, 2020-04-18:
@WillD7545: Sorry, my bad, I migrated this target without having access to an actual board. Fixed in betaflight/unified-targets#244.

WillD7545, 2020-04-18:
Thank you @Asizon and @mikeller. I must be the only one with one of these Nanos. 😝

closed 2020-04-19 by mikeller