betaflight/betaflight #9713: Switching profiles with/without dynamic idle bumps up the idle throttle

rvdveen, 2020-04-17:
When switching between a profile with and a profile without dynamic idle the idle value seems to get stuck too high when arming on the profile without dynamic idle.

**Steps to reproduce the behavior:**

1. Setup a pid profile without and a profile with dynamic idle.
2. Arm on the profile without dynamic idle, idle seems alright
3. Switch profiles and arm with dynamic idle, idle seems alright (a tad lower, as it should be)
4. Switch profiles and arm without dynamic idle, idle is way too high causing the quad to sometimes lift one side and flip out.

**Expected behavior**
I expect idle to be the same as before when switching back to a profile without dynamic idle.

**Flight controller configuration**
```
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr 13 2020 / 03:33:34 (fde6c680b) MSP API: 1.43

# config: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF4
manufacturer_id CLRA
mcu_id 003b004b564b500420343750
signature 

# name: init

# resources
resource BEEPER 1 NONE

# timer
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B04 NONE
timer B08 NONE

# dma
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B04 NONE
dma pin B08 NONE

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
[... 474 lines cut ...]
set thr_mid = 0
set roll_rc_rate = 118
set pitch_rc_rate = 118
set yaw_rc_rate = 140
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 52
set tpa_rate = 70
set tpa_breakpoint = 1350

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save
# 
```

etracer65, 2020-04-18:
Try this to see if it resolves the problem. Test initially with PROPS REMOVED.

[betaflight_4.2.0_STM32F405_9ef77df.hex.zip](https://github.com/betaflight/betaflight/files/4495713/betaflight_4.2.0_STM32F405_9ef77df.hex.zip)

rvdveen, 2020-04-18:
> Try this to see if it resolves the problem. Test initially with PROPS REMOVED.
> 
> [betaflight_4.2.0_STM32F405_9ef77df.hex.zip](https://github.com/betaflight/betaflight/files/4495713/betaflight_4.2.0_STM32F405_9ef77df.hex.zip)

Did back to back tests with the same diff, problem's gone with that build!

rvdveen, 2020-04-18:
Hmm so the idle throttle doesn't change with idle_min_rpm anymore but it does still happen when switching between profiles with ff_interpolate_sp = AVERAGED_2 versus AVERAGED_3

rvdveen, 2020-04-18:
Here's a video showing the problem. I'm switching to the following profiles (without airmode, doesn't seem to matter): 0,1,2,1,2,1

The problem only occurs the second time I'm switching to profile 1, when switching away from the profile with ff_interpolate_sp = AVERAGED_3

https://photos.app.goo.gl/oMV5grYsgDvQGAmeA

The diff for the video:
```
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr 17 2020 / 20:02:54 (9ef77df) MSP API: 1.43

# config: YES
# board: manufacturer_id: CLRA, board_name: CLRACINGF4
# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF4
manufacturer_id CLRA
mcu_id 003b004b564b500420343750
signature 

# name: init

# resources
resource BEEPER 1 NONE

# timer
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B04 NONE
timer B08 NONE

# dma
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B04 NONE
dma pin B08 NONE

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# aux
aux 0 0 1 1900 2100 0 0
aux 1 1 2 1700 2100 0 0
aux 2 13 3 1300 1700 0 0
aux 3 28 2 900 1300 0 0
aux 4 35 3 1700 2100 0 0
aux 5 36 0 1900 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 800

# master
set acc_calibration = 28,57,-32,1
set rssi_channel = 16
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set vbat_scale = 111
set ibata_scale = 230
set beeper_frequency = 0
set small_angle = 180
set pid_process_denom = 2
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_over_cap = ON
set osd_rssi_alarm = 70
set osd_cap_alarm = 1150
set osd_vbat_pos = 11
set osd_rssi_pos = 2048
set osd_link_quality_pos = 2052
set osd_tim_1_pos = 471
set osd_tim_2_pos = 2455
set osd_flymode_pos = 26
set osd_vtx_channel_pos = 6
set osd_current_pos = 2432
set osd_mah_drawn_pos = 2304
set osd_craft_name_pos = 0
set osd_display_name_pos = 396
set osd_power_pos = 352
set osd_pidrate_profile_pos = 26
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2328
set osd_disarmed_pos = 2218
set osd_log_status_pos = 22
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_max_g_force = ON
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 4
set vtx_freq = 5732
set vtx_pit_mode_freq = 5000
set vcd_video_system = NTSC
set name = init

profile 0

profile 1

profile 2

# profile 2
set ff_interpolate_sp = AVERAGED_3

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

mikeller, 2020-04-18:
@rvdveen: How are you actually switching PID profiles? Switching profiles should result in a reboot, as there are still a number of things that are not properly re-initialised when switching PID profiles without a reboot.

etracer65, 2020-04-18:
If there is any issue with switching profiles with different `ff_interpolate_sp` settings that's a completely different and unrelated to dynamic idle. The test you performed is not really definitive as the PID controller was active and many different things could cause a different motor response. Try repeating the test while logging and see how the throttle setpoint trace behaves. 

Also `ff_interpolate_sp` doesn't affect the throttle channel, only R/P/Y. So I'm not saying there isn't a problem, but I don't think it's related to throttle or dynamic idle at all.

rvdveen, 2020-04-18:
> @rvdveen: How are you actually switching PID profiles? Switching profiles should result in a reboot, as there are still a number of things that are not properly re-initialised when switching PID profiles without a reboot.

I'm switching profiles using stick commands in the video, although yesterday the problem cropped up when switching pid profiles using the osd.

I wasn't aware a reboot was necessary, that solves this issue then!

It might be a idea to make the reboot mandatory when switching pid profiles. It's a bit confusing to have the stick command for quick switches available but resulting in the quad taking off on its own on the next arm.

mikeller, 2020-04-18:
@rvdveen: I might be wrong here, but to my recollection changing PID profiles with stick commands results in a reboot - and in this case changing them in OSD _should_ require a reboot too.

@etracer65: Did we ever make PID profiles safe to be switched without a reboot, and remove the reboot when switching with stick commands?

etracer65, 2020-04-18:
@mikeller No, PID profile switching never required a reboot and shouldn't require one now. Not saying there aren't bugs that should be fixed (like the dynamic idle problem this issue is originally about), but it's designed to work without a reboot.

@rvdveen This secondary issue you're reporting regarding `ff_interpolate_sp` I think is a coincidence and related to your setup rather than a bug. If you're using stick commands to change profiles I would suggest it's likely one of your sticks is not completely centering. So after you use the sticks to change the profile one or more of the channels is maybe a microsecond or so from centered. When armed this would cause a small response fro the PID controller as observed as you would be unintentionally commanding a tiny amount of rotation.

I would remove the `ff_interpolate_sp` setting from profile 2 and retest to confirm that this is not related. You could also try adding deadband to your channels (Receiver tab) to see if that helps.

mikeller, 2020-04-18:
@etracer65: 

> No, PID profile switching never required a reboot and shouldn't require one now.

Yes, you are right - the original code did not do a reboot, but it stored the default PID profile into EEPROM, and then did a re-read of all parameters from EEPROM, and a re-init of everything.

rvdveen, 2020-04-18:
I've retested with some clean profiles, and again the issue only crops up after switching from a profile with AVERAGED_3 to a profile with the default AVERAGED_2.

Switching between empty profile doesn't cause the issue, nor does switching between dyn_lpf_dterm_curve_expo settings (I picked a random setting to test).

Attached are the logs with the profiles used in the names. The last log (`LOG00011_pro_1.BFL`) is a log with some movement (props off, just moving the quad around) and switching airmode on/off to test wether it was just some iterm building up or something.

The logs have RPM_FILTER as debug mode, weirdly only motors 1 and 3 spin up higher than necessary.

The amount they spin up after switching profiles seems kinda random, sometimes it's higher sometimes it's lower.

Profile 2 is the profile with the AVERAGED_3 setting, `LOG0006_pro_1.BFL` is the first log with issues.

[rpm_logs.zip](https://github.com/betaflight/betaflight/files/4496844/rpm_logs.zip)

etracer65, 2020-04-18:
Sorry, can't see anything in the logs other than a little bit of gyro noise induced PID activity (probably due to vibration from the motors) driving some motor response and a little jitter on your RC channels.

I tested switching back and forth between profiles that did/didn't have `ff_interpolate_sp = AVERAGED_3` and saw no effect. Idle RPM for all motors was identical in every case.

rvdveen, 2020-04-18:
Hm that's weird, in my explorer motors 1 and 3 are shown as being driven 4% with airmode off, jumping to 8% with airmode on in log 11 while motors 2 and 4 stay at 0%. Any ideas what other settings might be causing this?

I'll test the other quad tomorrow, and maybe try setting this one up from scratch without the remapping for non-bitbanged bidir rpm, maybe that's causing something funky.

etracer65, 2020-04-18:
> Hm that's weird, in my explorer motors 1 and 3 are shown as being driven 4% with airmode off, jumping to 8% with airmode on in log 11 while motors 2 and 4 stay at 0%.

PID response to either commanded inputs (jitter from channels) or gyro noise. For example if the pitch gyro sees a little bit of rate from noise it will command the front or back (1 & 3) motors to respond.

rvdveen, 2020-04-18:
> PID response to either commanded inputs (jitter from channels) or gyro noise. For example if the pitch gyro sees a little bit of rate from noise it will command the front or back (1 & 3) motors to [...]

I don't think it's jitter; rc command, gyros and pids are all 0 when it happens but the rear motors are stuck at ~8% (4% before airmode was switched on)
[image]

etracer65, 2020-04-18:
Gyro, PID and rcCommand are all floating point numbers. The blackbox log reports truncated integers. So if the pitch gyro was reporting 0.9 deg/sec the PID controller would respond to it but the log would show 0 deg/sec. Try setting your PID gains (P, I, D and F) for each axis to 0. Also set anti-gravity gain to 0. This will rule out a PID response.

And just to reiterate: unable to replicate.

rvdveen, 2020-04-18:
> Gyro, PID and rcCommand are all floating point numbers. The blackbox log reports truncated integers. So if the pitch gyro was reporting 0.9 deg/sec the PID controller would respond to it but the log [...]

Good point, I'll try that and test the other quad tomorrow. Gotta be something causing this :/ It's too consistent to chaulk this up to a funky gyro signal or something.

rvdveen, 2020-04-20:
Thanks for the debugging tips @etracer65! It would occur only with ff at defaults but all other pids at 1, with some ff caused by jitter when switching profiles.

PR #9719 (2020-04-20): Fix dynamic idle safety issues
Fixes #9713 

Fixes a problem with dynamic idle not checking for bidirectional DSHOT being enabled. So the code would run but have no RPM data available leading to elevated motor idle.

Fixes a problem with dynamic idle not resetting accumulated static variables that continue to be used when dynamic idle is switched off via a PID profile change. Depending on the state of the variables this could cause excessive motor idle speed. In some cases enough to cause the quad to spontaneously take off on arming or be unable to descend in flight.

Fixes a problem with dyna [... cut]
closed 2020-04-20 by rvdveen