betaflight/betaflight #9721: Hex X mix does not work

jlettow01, 2020-04-19:
After applying the Hex X mix the BOOT_GRACE arming disable flag never goes away

No matter what settings I change, the error still occurs

```
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.5 Mar 16 2020 / 05:19:58 (d4e74e39c) MSP API: 1.42
# manufacturer_id: CLRA   board_name: CLRACINGF4   custom defaults: YES

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id CLRA

# resources
resource BEEPER 1 NONE
resource MOTOR 6 A00
resource LED_STRIP 1 NONE
resource SERIAL_TX 4 NONE

# mixer
mixer HEX6X

# feature
feature -RX_PARALLEL_PWM
feature -OSD
feature RX_SERIAL
feature RSSI_ADC

# beacon
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 3 1700 2100 0 0
aux 4 35 2 1700 2100 0 0
aux 5 47 1 900 1300 0 0

# master
set acc_calibration = -130,96,-46
set serialrx_provider = SBUS
set min_throttle = 1070
set dshot_idle_value = 200
set use_unsynced_pwm = OFF
set motor_pwm_protocol = OFF
set motor_pwm_rate = 480
set beeper_frequency = 0
set small_angle = 180
set pid_process_denom = 16
set pwr_on_arm_grace = 1

profile 0

rateprofile 0

# end the command batch
batch end
```

**Setup / Versions**
 - bardwell f4 v2 from racedayquads
 - BF 4.1.5 and BF 4.1.3
 - motor5 to led strip and motor6 to rx4

**Additional context**
this flight controller only came with 5 motor outputs. I did all of the steps to add one more and remapped motor 5. I armed it once on BF 4.1.3 but it never let me arm after that. I have tried other pads such as tx6 for motor 6 but none of them fix the issue.

etracer65, 2020-04-19:
Hex mix works just fine. You just don't have 6 working motors.

jlettow01, 2020-04-19:
what other information do you need to know?

etracer65, 2020-04-19:
As requested in the issue template:

> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
>

Until you have 6 working motors (test in the motors tab without props) then you can't select a hex mixer.

jlettow01, 2020-04-19:
Thank you for your help and sorry for labeling this as a bug.

closed 2020-04-19 by jlettow01