betaflight/betaflight #9722: manufacturer-ID: FLYCOLOR

tobeornottobe81, 2020-04-20:
Hello, we would like to add a new manufacturer-ID: FLYCOLOR 
in betaflight Configurator
Our website is www.flycolor.net
My email is: 2851527075@qq.com
Thank you!

mikeller, 2020-04-20:
@tobeornottobe81: Am I right to assume that you will be the maintainer for the Flycolor targets?

tobeornottobe81, 2020-04-21:
You're right. Now I want to upload a flight control project based on stm32f722 chip of betaflight. But in the early stage of this process, I ran into a bit of trouble. According to the upload file guidelines, I found that there was no manufacturer ID, so I sent this problem in betaflight, hoping to solve it. thank you.

tobeornottobe81, 2020-04-21:
I follow  "How to create a unified target configuration"
Enter dump hardware to copy and modify the generated content as prompted. This file looks like that generated by other manufacturers
The file < manufacturer? ID > - < board? Name >. Config is similar. But I went to the last step, diff all bare
However, the generated files are different from those of other manufacturers. So the question is, which file should I upload? I'm looking forward to solving the problem. Thank you

Asizon, 2020-04-21:
For dump hardware use your current target, then flash your MCU target STM32xxxx, paste the dump hardware and continues steps finishing in diff all bare

closed 2020-04-21 by mikeller