betaflight/betaflight #9725: Geprc GEPRC SPAN-F7-BT-HD Bluetooth

TomAlperin, 2020-04-21:
**Describe the bug**
Geprc GEPRC SPAN-F7-BT-HD bluetooth is not working with the Speedy Bee app. It is possible to set a password, reset the password, and change the bluetooth name, but the connection times out when attempting to connect to change other settings. This was working properly with version previous to 4.1.5.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade flight controller to v4.1.5.
2. Attempt to connect using Speedy Bee app. 

**Expected behavior**
I would expect the app to connect as it previously did and allow changing of flight controller settings.

**Flight controller configuration**
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.5 Mar 16 2020 / 05:21:46 (d4e74e39c) MSP API: 1.42
# manufacturer_id: EXUA   board_name: EXF722DUAL   custom defaults: YES

# start the command batch
batch start

board_name EXF722DUAL
manufacturer_id EXUA

# name: Mark 4 HD

# feature
feature -SOFTSERIAL
feature ESC_SENSOR

# serial
serial 0 1 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 1 19200 57600 0 115200

# aux
aux 0 0 0 1350 2100 0 0
aux 1 1 2 1800 2100 0 0
aux 2 13 3 1300 2100 0 0
aux 3 26 0 1700 2100 0 0
aux 4 35 1 1800 2100 0 0
aux 5 36 3 1300 2100 0 0

# rxfail
rxfail 6 s 1000
rxfail 7 s 1000

# master
set gyro_lowpass2_hz = 400
set dyn_notch_range = AUTO
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 320
set dyn_lpf_gyro_max_hz = 800
set acc_calibration = 34,-17,-4
set align_mag = CW180
set mag_align_yaw = 1800
set mag_hardware = NONE
set min_check = 1010
set max_check = 1990
set sbus_baud_fast = ON
set dshot_bidir = ON
set bat_capacity = 1500
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set deadband = 3
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_cap_alarm = 2000
set osd_vbat_pos = 450
set osd_tim_2_pos = 298
set osd_flymode_pos = 14346
set osd_current_pos = 458
set osd_mah_drawn_pos = 14720
set osd_craft_name_pos = 14355
set osd_gps_speed_pos = 289
set osd_gps_lon_pos = 16
set osd_gps_lat_pos = 3
set osd_gps_sats_pos = 14
set osd_home_dir_pos = 225
set osd_home_dist_pos = 257
set osd_altitude_pos = 14656
set osd_warnings_pos = 457
set osd_avg_cell_voltage_pos = 14753
set osd_battery_usage_pos = 14825
set osd_esc_tmp_pos = 14689
set max7456_spi_bus = 0
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
set gyro_2_align_yaw = 900
set name = Mark 4 HD

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 112
set dyn_lpf_dterm_max_hz = 272
set dterm_lowpass2_hz = 240
set idle_min_rpm = 20

rateprofile 2

# rateprofile 2
set roll_rc_rate = 155
set pitch_rc_rate = 155
set yaw_rc_rate = 120
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: CAMERA_CONTROL
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: GYRO_EXTI
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: PINIO 3
B09: FLASH_CS
B10: I2C_SCL 2
B11: SERIAL_RX 3
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: GYRO_CS 2
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 3
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: PINIO 1
C14: PINIO 2
C15: BEEPER
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
**Setup / Versions**
 - Geprc GEPRC SPAN-F7-BT-HD Bought from Geprc;
 - DJI HD FPV system:
  Date: 2020.04.13
  Air Unit Firmware: v01.00.05.00
  Goggles Firmware: v01.00.05.00
  Remote Controller Firmware: v01.00.05.00;

 - Air unit is connected using wire harness supplied with stack.

**Additional context**
Bluetooth was configurable on previous version of Betaflight. It is possible to change settings on the bluetooth module, for example, password and device name. it is not possible to connect via bluetooth to change flight controller settings.

Steps to enable bluetooth were performed after flashing the latest firmware.
```
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
resource PINIO 1 C13
save
```

Asizon, 2020-04-21:
I'd like to tell you to use the official Betaflight app, but it's not fully implemented yet. We don't support Speedybee app sorry

TomAlperin, 2020-04-21:
Then why does this closed issue #9006 exist?

Asizon, 2020-04-21:
@TomAlperin this issue is refering to the flight controller board SPEEDYBEE F4

etracer65, 2020-04-21:
@TomAlperin The SpeedyBee app is produced by the "SpeedyBee" company and we have nothing to do with it. Please contact them or possibly GEPRC who manufacturer the bluetooth module.

TomAlperin, 2020-04-21:
This isn’t an issue with the app. The issue is with the connection of the Bluetooth module connection similar to the issue mentioned above that was about the Speedy Bee flight controller, but specifically about the Bluetooth connection on that controller.

etracer65, 2020-04-21:
Then contact the bluetooth module manufacturer for documentation and support about how to use their module.

mikeller, 2020-04-22:
@TomAlperin: As you were told before, go and ask the supplier of the hardware to make this work with Betaflight 4.1.5 then. This manufacturer does not collaborate with us, or even supply us engineering samples of their hardware. From the look of it they do not even bother to provide their customers with a proper target to make their hardware work, requiring the use of another manufacturer's target by changing some settings. Closing this.

closed 2020-04-22 by mikeller