betaflight/betaflight #9727: manufacturer ID request - Hobbywing

Project399, 2020-04-22:
Requesting add of a new manufacturer ID:
To help customers have an easier time with setup, we would like to add a Hobbywing manufacturer-id.

Manufacturer: HOBBYWING
Company Website: http://hobbywing.com/

Thank you

# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/);
5. Remove this Text :).

mikeller, 2020-04-22:
For some reason the website does not seem to be working. Can you please get this fixed first?

Project399, 2020-04-22:
The website works on my end, just a bit slow to load  today it seems for some reason.

mikeller, 2020-04-25:
@Project399: I stand corrected, the website does work - I had given up after a minute or so. Maybe you want to replace that 56k modem with something a bit faster? ;-)

Jokes aside, am I right to assume that you will be the maintainer of the Hobbywing targets?

Project399, 2020-04-25:
Haha I have made them aware, and yes that is correct. Thank you

closed 2020-04-26 by mikeller