betaflight/betaflight #9729: Arming disable flag RPMFILTER not showing

TheIsotopes, 2020-04-23:
**Describe the bug**
If i'm using `dshot_bidir = ON` on FC with default `dshot_burst = ON` and `dshot_bitbang = AUTO` the Arming disable flag `RPMFILTER` is not showing but everything works perfectly.

When i using `dshot_burst = OFF` and `dshot_bitbang = OFF` with `dshot_bidir = ON` the  Arming disable flag `RPMFILTER` is appeared in BFC but nothing else has changed, everything works perfectly.

**Flight controller configuration**

diff output:
```
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Apr 22 2020 / 12:51:00 (bf9cc615c) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722MINI, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

# name: HEXAMOVR

# mixer
mixer HEX6X

# feature
feature -TELEMETRY
feature -OSD
feature LED_STRIP

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# map
map TAER1234

# led
led 0 9,2::AW:0
led 1 9,3::AW:0
led 2 9,4::AW:0
led 3 9,5::AW:0
led 4 6,5::AW:0
led 5 6,4::AW:0
led 6 6,3::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1600 2000 0 0
aux 1 13 1 1000 1200 0 0
aux 2 26 0 1800 2000 0 0
aux 3 31 0 1200 1400 0 0
aux 4 36 1 1800 2000 0 0

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 80
set dyn_notch_max_hz = 350
set acc_hardware = NONE
set rc_smoothing_auto_smoothness = 5
set serialrx_provider = SUMD
set dshot_bidir = ON
set bat_capacity = 1550
set use_cbat_alerts = ON
set cbat_alert_percent = 20
set vbat_duration_for_warning = 50
set vbat_duration_for_critical = 20
set gyro_rpm_notch_q = 700
set name = HEXAMOVR

# profile 0
set iterm_relax_cutoff = 30
set throttle_boost = 7
set throttle_boost_cutoff = 25
set ff_interpolate_sp = ON
set ff_smooth_factor = 20

# rateprofile 0
set rates_type = QUICK
set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 140
set roll_srate = 126
set pitch_srate = 126
set yaw_srate = 99

# end the command batch
batch end
```
resource show all output:
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: MOTOR 5
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: MOTOR 6
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: LED_STRIP
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
found on MATEKF722MINI and MATEKF405MINI

etracer65, 2020-04-23:
The Arming disable flag RPMFILTER means that RPM filter is enabled but RPM telemetry is not being received from one or more ESCs. Check the Motors tab to view the error percentages.

TheIsotopes, 2020-04-23:
on both variants the motors are spinning correct and also the RPM telemetry ist showing in cli `dshottelemertryinfo`. flying with both variants have also no problems.
only one variant shows he Arming disable flag `RPMFILTER` and the other not.

etracer65, 2020-04-23:
How can you fly if the arming disable flag is set? That's the point of arming disable - it prevents arming.

TheIsotopes, 2020-04-23:
is that really hard to understand what i mean? new trial.

first setup ->` dshot_bidir = ON` , `dshot_burst = ON` and `dshot_bitbang = AUTO` -> without battery **no** Arming disable flag `RPMFILTER` is seen on BFC

second setup -> ` dshot_bidir = ON` , `dshot_burst = OFF` and `dshot_bitbang = OFF` -> without  battery Arming disable flag `RPMFILTER` is appearing on BFC

both setup's flying without any problems with rpm filter enabled, `dshottelemeryinfo` in cli also without any problems

TheIsotopes, 2020-04-23:
in a nutshell, actually in both cases arming disabled flag `RPMFILTER` should be displayed if no battery is connected!? or did I misunderstand something?

Asizon, 2020-04-23:
Effectively appears rpmfilter flag, in the case of having dshot_burst on, dshot bidir remains disabled:
`# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.31V, Core temp=40degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3960, Max available config: 16384
Gyros detected: gyro 2
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
OSD: MAX7456
System Uptime: 22 seconds, Current Time: 2020-04-23T15:10:36.510+00:00
CPU:4%, cycle time: 112, GYRO rate: 8928, RX rate: 33, System rate: 10
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 6
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP RPMFILTER`

TheIsotopes, 2020-04-23:
first case:
```
# get dshot
dshot_idle_value = 550
Allowed range: 0 - 2000

dshot_burst = ON
Allowed values: OFF, ON, AUTO

dshot_bidir = ON
Allowed values: OFF, ON
Default value: OFF

dshot_bitbang = AUTO
Allowed values: OFF, ON, AUTO

dshot_bitbang_timer = AUTO
Allowed values: AUTO, TIM1, TIM8

beeper_dshot_beacon_tone = 1
Allowed range: 1 - 5

# status
MCU F722 Clock=216MHz, Vref=3.29V, Core temp=40degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3962, Max available config: 16384
Gyros detected: gyro 1, gyro 2
GYRO=MPU6000
OSD: NONE
System Uptime: 41 seconds, Current Time: 2020-04-23T15:16:21.651+00:00
CPU:2%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 4 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS NOPREARM CLI MSP

```

second case:
```
# get dshot
dshot_idle_value = 550
Allowed range: 0 - 2000

dshot_burst = OFF
Allowed values: OFF, ON, AUTO
Default value: ON

dshot_bidir = ON
Allowed values: OFF, ON
Default value: OFF

dshot_bitbang = OFF
Allowed values: OFF, ON, AUTO
Default value: AUTO

dshot_bitbang_timer = AUTO
Allowed values: AUTO, TIM1, TIM8

beeper_dshot_beacon_tone = 1
Allowed range: 1 - 5

# status
MCU F722 Clock=216MHz, Vref=3.28V, Core temp=49degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3962, Max available config: 16384
Gyros detected: gyro 1, gyro 2
GYRO=MPU6000
OSD: NONE
System Uptime: 22 seconds, Current Time: 2020-04-23T15:18:58.225+00:00
CPU:3%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 4 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS NOPREARM CLI MSP RPMFILTER

```

TheIsotopes, 2020-04-23:
The question of exactly what the correct settings are is because the wiki proposes two different setups.

Fist wiki entry [Bidirectional-DSHOT-and-RPM-Filter](https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter) says this config is correct (corresponds to the second case):
```
set dshot_burst = OFF
set dshot_bidir = ON
set dshot_bitbang = OFF
```
The second wiki entry [Simpler RPM Filter Configuration](https://github.com/betaflight/betaflight/wiki/4.1-Tuning-Notes#simpler-rpm-filter-configuration) says `RPM telemetry will be automatically activated when dshot_bidir is enabled, and this can be done in the 10.6 Configurator.`
imho defaults config with `dshot_bidir = ON` (same with the first case):
```
dshot_burst = ON
dshot_bidir = ON
dshot_bitbang = AUTO
```

etracer65, 2020-04-23:
I found the problem. The detection of valid telemetry data is not working properly for bitbang (it always returns true). Working on a fix.

etracer65, 2020-04-24:
Should be fixed in #9730. If you want to test try this:

[betaflight_4.2.0_STM32F7X2_62adae7.hex.zip](https://github.com/betaflight/betaflight/files/4525920/betaflight_4.2.0_STM32F7X2_62adae7.hex.zip)

TheIsotopes, 2020-04-24:
working now ... @etracer65 thx for fixing 👍 

```
# get dshot
dshot_idle_value = 550
Allowed range: 0 - 2000

dshot_burst = ON
Allowed values: OFF, ON, AUTO

dshot_bidir = ON
Allowed values: OFF, ON
Default value: OFF

dshot_bitbang = AUTO
Allowed values: OFF, ON, AUTO

dshot_bitbang_timer = AUTO
Allowed values: AUTO, TIM1, TIM8

beeper_dshot_beacon_tone = 1
Allowed range: 1 - 5

# status
MCU F722 Clock=216MHz, Vref=3.26V, Core temp=43degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3962, Max available config: 16384
Gyros detected: gyro 1, gyro 2
GYRO=MPU6000
OSD: NONE
System Uptime: 31 seconds, Current Time: 2020-04-24T06:34:09.049+00:00
CPU:1%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 4 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Not configured
Arming disable flags: RXLOSS NOPREARM CLI MSP RPMFILTER
```

PR #9730 (2020-04-25): Fix ARMING_DISABLED_RPMFILTER when DSHOT bitbang is enabled
Fixes #9729 

The function that checks for every motor having valid telemetry broke with the implementation of bitbang. Effectively the motor count was zero and this caused the logic to fall through with a false-positive. The check relied on a local motor count that was set when the PWM dshot was initialized. When bitbang is active this initialization no longer runs and the motor count was defaulting to zero. This was missed in the bitbang implementation.

Fixed to get the motor count from a common source initialized by both timer and [... cut]
closed 2020-04-25 by mikeller