betaflight/betaflight #9731: DJI FPV Air Unit + Kakute F7 HDV FC + Beta Flight + Tekko32 F3 HDV 4-in-1 ESC

iskymaster, 2020-04-24:
I am experiencing a strange anomaly and I have sent a support request to Holybro as well.

I am not sure if its Holybro FC problem, or beta flight problem?

I got the Holybro GPS all wired up and its showing up in beta Flight. 

**ISSUES:**

1. Once the Motors are Armed and I am in ANGLE Mode. if I increase the throttle the motors starts to spin faster. But when I bring the throttle down, the motors continue to spin at the same speed or they appear they do. Doesn't look like throttle down has any effect on it. Why is that? there are no props on for this test.

I am using DSHOT600 and also tried OneShot125

2. I tried with props on and the drone went crazy. I was holding the drone in my hand (I had a feeling there is some problem) and the motors started spin wildly at different speeds and even went to full speed. The throttle had no effect on it. 
 
Does this has anything to do with GPS and Mode settings or there is something wrong here?

Thanks for your help.

[image]
[image]
[image]
[image]

sjbshrss, 2020-04-24:
It's normal cause the Airmode is activated，try to fly without holding the drone.

etracer65, 2020-04-24:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook https://www.facebook.com/groups/betaflightgroup/).

closed 2020-04-24 by etracer65