betaflight/betaflight #9736: SUMD receiver on UART6 not working

noah95, 2020-04-26:
**Describe the bug**
SUMD serial receiver on UART6 not working after 4.1 but works on RX1

**To Reproduce**
Steps to reproduce the behavior:
Flash BFL 4.1.6, connect SUMD receiver to UART6 Rx, receiver signal not working
Flash BFL 3.5.6, connect SUMD receiver to UART6 Rx, receiver signal is working

With BFL 4.1.6 receiver on UART1 Rx, receiver is working

**Flight controller configuration**
BETAFLIGHTF4, only thing changed is receiver serial SUMD and serial port configuration

**Setup / Versions**
Graupner GR-12L receiver

**Additional context**
Might be related to closed #7849

Powerup with 3.5.6
[image]

Powerup with 4.1.6
[image]

My guess: Is there a pull-up not turning on on pin C07? (Rx6)

etracer65, 2020-04-26:
Put a pull-up resistor on RX6 (maybe 10K) and see if that fixes it.

noah95, 2020-04-26:
physical 10k pull up on rx6 doesn't work. 

From this image i see the inverter logic:
https://static.rcgroups.net/forums/attachments/5/9/3/2/6/3/a10375788-166-BFF4_WWS_v1_0.jpg

How can that work if i connect my non-inverted receiver to the output of an inverter? That part should have a tri-state output controllable by the MCU, shouldn't it?

noah95, 2020-04-26:
Is the BFF4 target incomplete and a change in unused GPIOs has caused this inverter to not switch off if unused? Although if that's the case, i can't believe i am the first noticing it..

etracer65, 2020-04-26:
The earlier problems you're referencing were the result of cloned boards not properly setting the pull-up/down state on their inverter control pin and relying on the accidental state of the MCU pin. Changes in initialization and restructuring caused the "accidental" pin state to not be adequate to control the bad design. If that's related to your problem you could try:
```
resource PULLUP 1 <pin>
```
Where `<pin>` is the gpio pin associated with the inverter control. `PULLDOWN` can be used instead if the logic needs that.

noah95, 2020-04-26:
I can't find the pin that controls the inverter. It should be defined in the `target.h` file, right?

mikeller, 2020-04-26:
@noah95:

> I can't find the pin that controls the inverter. It should be defined in the target.h file, right?

No, since this is not part of the official board specification this is not an actively controlled pin, just a case of the designer of this board abusing an unused MCU pin that happened to be at a given level for some firmware version to save a few cents on a resistor.
Try all unused pins, or ask the manufacturer / supplier of the board for a schematic.

noah95, 2020-04-27:
Oh @mikeller you misunderstood me there. I have the original board, not some clone. There is even FPVModel engraved in the OSD chip, so I am 100% certain that it is genuine. 

I contacted the manufacturer nontheless for a schematic.

mikeller, 2020-04-27:
@noah95: Ok, I see - then there is a chance that we will be able to get a schematic and solve this that way. I would not start desoldering - this will still be just guesswork without the schematic.

jflyper, 2020-04-27:
@noah95 Lemme see how your board look like.

noah95, 2020-04-27:
Let me know if you need a cleaner shot. The SUMD pin is the one with the soldered on 100mil header top. The bottom of the two is for sbus. My guess is that the 8pin package next to it is the inverter. 

[image]

jflyper, 2020-04-28:
@noah95 

Try this

```
resource pulldown b8
```

mikeller, 2020-04-28:
@jflyper:

> resource pulldown 1 b8

Ha, good idea to try PPM. Do you actually have a schematic for the board?

noah95, 2020-04-28:
> @noah95
> 
> Try this
> 
> ```
> resource pulldown 1 b8
> ```

Works! This pin is the `ESCSERIAL_TIMER_TX_PIN`, coud @jflyper elaborate how this is hooked up in HW?

Could you give me a hint on where this could be fixed in the target configuration? 

[image]

jflyper, 2020-04-28:
> From this image i see the inverter logic:
> https://static.rcgroups.net/forums/attachments/5/9/3/2/6/3/a10375788-166-BFF4_WWS_v1_0.jpg
> 
> How can that work if i connect my non-inverted receiver to the output of an inverter? That part should have a tri-state output controllable by the MCU, shouldn't it?

The inverter for SBUS in this figure is conceptual one and not a logic gate. It is composed of a transistor and a couple resistors that works as an inverter. Something like this:

[image]

This circuit was first introduced by Omnibus F4 and is pretty much common among F4 boards. The design that shares SBUS and PPM inputs on the same pin is also introduced by Omnibus F4 and is used in many boards also. This worked in most configurations until we decided to explicitly pull-up unused pins in #9376 (and #9398). With this change, PB11 will be pulled up which causes the transistor to turn on, which in turn hard pull UART6_RX down. The `resource pulldown 1 b11` will keep the transistor off and from UART6_RX from being hard pulled down.

noah95, 2020-04-28:
Interesting... Would this PR fix this: https://github.com/betaflight/unified-targets/pull/257

mikeller, 2020-04-28:
@jflyper: Interesting, I did not know that this hack was introduced by the OMNIBUS - it is otherwise a solid design.

closed 2020-06-06 by stale[bot]