betaflight/betaflight #9737: Configurator 10.6.0 not applying custom defaults for BF 4.1.6 to Unified SPRacing F4EVO

Poitivin, 2020-04-26:
Hello
I've flashed the "betaflight_4.1.6_STM32F405.hex" for SPRacingF4 EVO (SPRO), however after flash, configurator 10.6.0 is not applying custom defaults for BF 4.1.6 to this Unified SPRacing F4EVO.
Any help appreciated

# diff all
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F405 (S405) 4.1.6 Apr 25 2020 / 05:11:18 (283bda8bf) MSP API: 1.42
# board_name: SPRACINGF4EVO   custom defaults: NO

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SPRACINGF4EVO
mcu_id 003100553036510135353735
signature 

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

Poitivin, 2020-04-26:
Am closing this "issue" as I found this is related with a driver (USB) rather with BF Configurator.

None of my PCs were able to set the DFU mode in these boards (only). I have a Kakute and this works ok.
So, I ended loading the unified *.hex with the STM32 ST-LINK Utility as that was able to find the DFU mode and load the hex, however as this was the first time I did not loaded the legacy *.hex, I did not realized the flash of the unified must be done with the configurator so it loads the *.hex and also the specific board *.confg file.

Further testing showed me that in any machine with USB 3 (only) ports, the DFU in this board fails. Trying in a machine with USB 2, seems to work just fine from the BF Configurator.

Apologize for any confusion this may have caused and thanks for all efforts BF devs applied so far

closed 2020-04-26 by Poitivin

mikeller, 2020-04-26:
@Poitivin: Yes, flashing targets with ST-Link or other tools is not supported any more from 4.1 on. The code used to stitch the board-specific configuration into the generic STM32... target on the fly is part of Betaflight Configurator, making this the only tool that can flash fully functional Betaflight firmware (at least for the moment).

> Further testing showed me that in any machine with USB 3 (only) ports, the DFU in this board fails. Trying in a machine with USB 2, seems to work just fine from the BF Configurator.

This is interesting. What operating system / hardware is this happening on?

Poitivin, 2020-04-26:
Hello Michael and thanks for reaching out.
My any machine means at least 3 laptops I own all running Win10 (1909)
on same machines my KakuteF7 works like a charm, but the 4 SP Racing EVO F4 get into the issue that DFU simply does not load, however as these boards have a hardware button, if i boot with the button pushed, I go to DFU with no problem and can flash the unified hex and all works ok, so am not at this point this is USB 3 vs 2 or some other driver.
Any attempt to boot to DFU from the configurator, does not load any DFU driver. Device Manager does not show any attempt to go to DFU.
Am still trying to figure out why and specifically on these SPR EVO boards