betaflight/betaflight #9745: Y6 doesn't work well with Dshot

Wimleung, 2020-05-01:
FC: BeeRotor F4 mini
ESC: Dshot compatible
Firmware: tried 3.5.7/4.1.1/4.1.6
Issue: 
Works fine with oneshort & Y6/octo X8.
If I choose Dshot (150/300/600) under Y6, motors number 1/3/4/5 works fine, motors number 2&6 won't spin but just "singing". 

ESCs are all good(if change the mixer to X4, motor2# will spin). 

Plus: If I choose Dshot under ocoto X8, none motor will spin.

Wimleung, 2020-05-01:
@mikeller Sorry what information is required to resolve this issue? Thanks.

mikeller, 2020-05-01:
@Wimleung: The one that you were asked to provide when opening this issue.

jflyper, 2020-05-01:
@Wimleung Required information is described in the pre-formatted text that was available when you first opened this page (you have deleted it).

Wimleung, 2020-05-01:
@jflyper Thanks. Let me try to add all information, although I think the issue is quite clear.

jflyper, 2020-05-01:
Superficial issue may be clear, but there might be something underneath it. Moreover, readers are not aquinted with your hardware and software configs at all. The requested information is the minimal start point for trouble shooting.

Wimleung, 2020-05-01:
@jflyper Oh I See. I am preparing the infor needed. Thanks again.

Wimleung, 2020-05-01:
**Describe the bug**
I made a tiny Y6 (6 motors), when I ran it with ESC setting "oneshot 125" it was all good.
When I change the setting to "Dshot300", motor 2# & 6# don't spin. ESC are good. (Dshot150/600 tried with same result)

**To Reproduce**
Steps to reproduce the behavior: 
1. Flash ESCs with BLS16.7, set FC with "oneshot125", everything works well.
2. Flash ESCs with JESC 2.3 48KHz firmware, and set FC with "Dshot300" which is supported by the ESC firmware, only motor 1#3#4#5# will spin. Motors and ESCs were checked without any problem.
3. Plus: If then change the mixer to be "X4", motors 1-4 will spin.

**Expected behavior**
I jsut want to run the Y6 with Dshot 300, because the 48KHz ESC firmware is battery saving and can lead to a longer fly time.

**Flight controller configuration**

'''# version
# Betaflight / BEEROTORF4 (BRF4) 4.1.1 Nov 15 2019 / 12:12:02 (1e5e3d369) MSP API: 1.42

# start the command batch
batch start

board_name BEEROTORF4

# mixer
mixer Y6

# feature
feature -AIRMODE

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 255 900 1250 0 0
aux 2 2 255 1300 1700 0 0

# master
set motor_pwm_protocol = DSHOT300
set vbat_max_cell_voltage = 435
set osd_vbat_pos = 2497
set osd_tim_1_pos = 2103
set osd_tim_2_pos = 2135
set osd_flymode_pos = 2475
set osd_throttle_pos = 2337
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_mah_drawn_pos = 2400
set osd_warnings_pos = 14792
set osd_avg_cell_voltage_pos = 2465
set osd_disarmed_pos = 2441

profile 0

rateprofile 0

# end the command batch
batch end'''

'''# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 4
A01: MOTOR 3
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: BEEPER
B04: LED 1
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: SDCARD_DETECT
C04: FREE
C05: USB_DETECT
C06: MOTOR 5
C07: MOTOR 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: INVERTER 3
C15: INVERTER 2
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH2: MOTOR 1
TIM2:
    CH2: MOTOR 3
TIM3:
    CH1: MOTOR 5
TIM4: FREE
TIM5:
    CH1: MOTOR 4
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: MOTOR 6
    CH3: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 4
DMA1 Stream 3: FREE
DMA1 Stream 4: MOTOR 5
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: MOTOR 6
DMA2 Stream 3: FREE
DMA2 Stream 4: MOTOR 2
DMA2 Stream 5: FREE
DMA2 Stream 6: MOTOR 1
DMA2 Stream 7: FREE '''

**Setup / Versions**
 - BeeRotor F4 mini from RCTIMER
 - ESCs are EMAX Bullet. BLS 16.7 for oneshort125, JESC 2.3(48KHz) for Dshot 300;
 - Mixer is Y6, 6 ESC/Motor sets connnected to the 1#-6# of the 8 outputs.

**Additional context**
I even tried on another Octo X8 which runs well with oneshot125, when I set "Dshot300", none motor will spin, it looks like Dshot is toally not supported by X8 mixer mode? Do want it possible for a longer fly time. Thanks.

Wimleung, 2020-05-01:
Pictures of the Y6 and X8
[image]
[image]

jflyper, 2020-05-01:
Hmm... from what is given, it seems that motor 6 on TIM8_CH2 is playing some role here... Have to give the config to dev board. Will take a while (other projects are queued).

Wimleung, 2020-05-01:
@jflyper Thanks. Since it looks complicated, I'd rather wait until Y6 issue is resolved before I attach the X8 setup data here.

mikeller, 2020-05-03:
@Wimleung:

In order to be able to use 6 motors you need to set `dshot_burst = on` in CLI:

[image]

If you want to use this board for an octocopter with Dshot you will have to use the LED_STRIP pin instead of M8, use `resource motor 8 b8` to change the motor pin.

Wimleung, 2020-05-03:
@mikeller I happened to find the old firmware worked, so I tried to downgrade the firmware step by step from 4.1.1 until I found the 4.04 worked for both Y6 and X8 on PWM output 1-8(4.05 didn't). Please see the following for 4.04 Octo X8. 
'''# version
# Betaflight / BEEROTORF4 (BRF4) 4.0.4 Jun 30 2019 / 14:00:45 (f3a95efa3) MSP API: 1.41

# start the command batch
batch start

board_name BEEROTORF4
manufacturer_id 

# name

# resources

# timer

# dma

# mixer
mixer OCTOX8

# servo

# servo mix

# feature
feature -AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon

# map

# serial

# led
led 0 7,6::C:2
led 1 8,6::C:2

# color

# mode_color

# aux
aux 0 0 0 1775 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1800 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT300
set osd_vbat_pos = 2465
set osd_tim_1_pos = 2103
set osd_tim_2_pos = 2135
set osd_throttle_pos = 2177
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2400
set osd_warnings_pos = 14760
set osd_avg_cell_voltage_pos = 2433
set osd_disarmed_pos = 2441
set osd_esc_rpm_pos = 216

# profile
profile 0

set d_pitch = 33
set d_roll = 32

# rateprofile
rateprofile 0

# end the command batch
batch end'''

'''Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 4
A01: MOTOR 3
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: BEEPER
B04: LED 1
B05: MOTOR 7
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: MOTOR 8
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: ADC_CURR
C02: FREE
C03: FREE
C04: FREE
C05: USB_DETECT
C06: MOTOR 5
C07: MOTOR 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: INVERTER 3
C15: INVERTER 2
D02: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE'''

mikeller, 2020-05-04:
@Wimleung: Please see https://github.com/betaflight/betaflight/issues/9745#issuecomment-623104225.

Wimleung, 2020-05-04:
@mikeller firmware 4.04 works perfect for X8 and Y6 with Dshot, without the needs to use the LED output.

mikeller, 2020-05-04:
@Wimleung: Not sure what you want to tell me, but I have given you the instructions for how to get this to work with current firmware, and this can be closed.

Wimleung, 2020-05-04:
@mikeller Sorry I must have misunderstood you, I will have try on the latest firmware and report the result here. Thanks.