betaflight/betaflight #9748: VTX does not take vtx_power command after update to BF 4.2

JoushiChef, 2020-05-01:
**Describe the bug**
I have setup the video transmitter tables
https://i.imgur.com/txdNq9q.jpg
And I can use smartaudio through the OSD to change the vtx_power setting, as in when I check after changing it in the video transmitter tab or CLI, it will show the correct value that I set.
However, the VTX's bootsequence always shows maximum output (800mw, level 4 in my case 3 red blinks followed by 4 blue blinks)

**To Reproduce**
Whatever I do the VTX's bootsequence is always the same, showing maximum power output.

**Expected behavior**
When I set vtx_power to 1 I expect the VTX to be set to 25mw, and 2 and 3 accordingly to 200 and 500mw.

**Flight controller configuration**
diff
```
# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr 29 2020 / 08:08:49 (b480103d4) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4FW, version: b0bd8e54, date: 2019-10-20T13:05:07Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4FW
manufacturer_id AIRB
mcu_id 00450034574850122034344b
signature 

# beacon
beacon RX_SET

# serial
serial 0 0 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1750 2100 0 0
aux 1 2 1 900 1225 0 0
aux 2 13 2 1875 2075 0 0
aux 3 28 1 1350 2100 0 0
aux 4 35 3 1775 2075 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 13 23 27 29
vtxtable powerlabels 25 200 500 800

# master
set gyro_to_use = SECOND
set acc_calibration = -66,4,-491,1
set min_check = 1005
set max_check = 1995
set rssi_channel = 16
set serialrx_provider = SBUS
set blackbox_device = SERIAL
set motor_pwm_protocol = DSHOT600
set align_board_roll = 180
set osd_vbat_pos = 2433
set osd_rssi_pos = 2425
set osd_tim_2_pos = 2113
set osd_vtx_channel_pos = 2081
set osd_current_pos = 2466
set osd_mah_drawn_pos = 398
set osd_warnings_pos = 2409
set osd_disarmed_pos = 2474
set debug_mode = SMARTAUDIO
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 3
set vtx_freq = 5880
set gyro_1_align_yaw = 1800
set gyro_2_align_pitch = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 1

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 150
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 10
set yaw_srate = 60

rateprofile 1

# rateprofile 1
set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 5
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 75

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 1

# save configuration
save
# 
```
resource
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: MOTOR 3
A04: GYRO_CS 2
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED 1
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: BARO_CS
B04: BEEPER
B05: MOTOR 4
B06: FREE
B07: FREE
B08: FREE
B09: CAMERA_CONTROL
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: PREINIT 1
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH4: MOTOR 3
TIM3:
    CH2: MOTOR 4
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: MOTOR 4
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
Airbot Omnibus F4 V6 bought here
https://www.banggood.com/Original-Airbot-OMNIBUS-F4-V6-Flight-Controller-Typhoon32-V2_1-35A-Blheli_32-3-6S-Brushless-ESC-p-1366629.html?cur_warehouse=CN
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
VTX is a TBS Unify Pro 5g8 hv
https://www.team-blacksheep.com/tbs-unify-pro-5g8-manual.pdf

**Additional context**
I did not have this problem before the update, so I am very confident everything is wired up correctly.

etracer65, 2020-05-01:
The vtxtable you used looks incorrect. Follow the directions at the top of this page to determine the smartaudio version used by your vtx:

https://github.com/betaflight/betaflight/wiki/VTX-tables

I suspect it will be smartaudio 2.0.  Look further down the page and grab the appropriate vtxtable and load that. Using the wrong table will prevent switching power levels. Also follow the manual for your vtx to make sure it's unlocked.

JoushiChef, 2020-05-02:
> The vtxtable you used looks incorrect. Follow the directions at the top of this page to determine the smartaudio version used by your vtx:
>
> https://github.com/betaflight/betaflight/wiki/VTX-tables [...]

It was smart audio 2.0 and I used the corresponding file/table (vtx_table_smart_audio_2_0_us.json).
Why/where does it look incorrect?
The VTX is unlocked. I have also tried locking and then unlocking it again. No change in behavior.

etracer65, 2020-05-02:
Your `diff` does not show a SA2.0 table. The `vtxtable powervalue` settings are incorrect for SA2.0. Please reload the SA2.0 table.

JoushiChef, 2020-05-02:
Wait I thought I had to adjust the powerlevels? 
If I just load the table, I get the values "1" "2" "3" with labels up to 500mw. It says in betaflight to adjust the power values according to the manufacturer, and in the TBS unify manual it says
14dBm (25mW)
23dBm (200mW*)
27dBm (500mW*)
29dBm (800mW*)
Do I have to adjust those numbers somehow? Surely putting 1,2,3,4 for power values will not give me the correct mW strengths.

I just realized in the table I had 13dBm instead of 14, I changed that, but that didn't change the bevahior of the vtx (same problem still).
I also reloaded the SA2.0 table, no change in behavior.

etracer65, 2020-05-02:
SA2.0 uses power values like 0, 1, 2 - not dBm values. What the actual outputs correspond to depends on the vtx and how the manufacturer implemented it. The values are just placeholders and don't correspond to any calculated power level. 0 = lowest, 1 = next higher, 2 = next higher, and so on. You'll need to get the actual supported vales from TBS.

JoushiChef, 2020-05-02:
Shouldve watched that video I saw all the way to the end I guess...
Thank you!!

closed 2020-05-02 by JoushiChef