betaflight/betaflight #9749: iFlight SucceX-D F7 V2.1 Stack Firmware Issue

fuleinist, 2020-05-02:
**Describe the bug**
Firmware target IFLIGHT_F722_TWING(IFRC) is not working for iFlight SucceX-D F7 V2.1 Stack (DJI HD Version). Currently only working with EXF722DUAL(Legacy) 4.1.1

**To Reproduce**
Steps to reproduce the behaviour:
Flash iFlight SucceX-D F7 with  IFLIGHT_F722_TWING(IFRC) and any version

**Expected behaviour**
FC booting and working as normal

**Current behaviour**
FC auto-reboot failed after a flush, sensors are not connected after a manual reboot 

**Flight controller configuration**
```
# diff

# version
# Betaflight / EXF722DUAL (EX7P) 4.1.1 Nov 15 2019 / 12:19:23 (1e5e3d369) MSP API: 1.42

# start the command batch
batch start

board_name EXF722DUAL

# name: DJI HD 7

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# feature
feature -RSSI_ADC
feature -AIRMODE
feature -ANTI_GRAVITY
feature SOFTSERIAL
feature GPS

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 1 115200 57600 0 115200
serial 2 2 115200 9600 0 115200
serial 30 32 115200 57600 0 115200

# led
led 0 4,15::CTOB:0
led 1 4,13::CTOB:0
led 2 11,15::CTOB:0
led 3 11,13::CTOB:0

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 27 3 1700 2100 0 0
aux 4 46 3 1300 1700 0 0
aux 5 13 3 1700 2100 0 0
aux 6 28 1 1700 2100 0 0
aux 7 35 2 1700 2100 0 0

# master
set gyro_lowpass2_hz = 188
set dyn_notch_range = LOW
set dyn_lpf_gyro_min_hz = 150
set dyn_lpf_gyro_max_hz = 375
set acc_trim_roll = -2
set acc_calibration = 77,10,14
set rssi_channel = 16
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 20
set failsafe_throttle = 1150
set failsafe_procedure = GPS-RESCUE
set vbat_scale = 112
set ibata_scale = 110
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_throttle_hover = 1350
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 2
set osd_vbat_pos = 2093
set osd_rssi_pos = 2081
set osd_link_quality_pos = 2084
set osd_tim_1_pos = 2455
set osd_tim_2_pos = 2487
set osd_flymode_pos = 2412
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2176
set osd_mah_drawn_pos = 2210
set osd_gps_lon_pos = 2433
set osd_gps_lat_pos = 2465
set osd_gps_sats_pos = 2105
set osd_home_dir_pos = 2348
set osd_home_dist_pos = 2349
set osd_altitude_pos = 2134
set osd_core_temp_pos = 2113
set name = DJI HD 7

profile 0

# profile 0
set p_pitch = 55
set i_pitch = 99
set d_pitch = 42
set f_pitch = 105
set p_roll = 51
set i_roll = 94
set d_roll = 39
set f_roll = 99
set p_yaw = 33
set i_yaw = 99
set f_yaw = 99
set d_min_roll = 22
set d_min_pitch = 24

rateprofile 0

# rateprofile 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

# end the command batch
batch end
```

**Additional context**
Add any other context about the problem here.

mikeller, 2020-05-02:
@fuleinist: Just tested this board with 4.2.0-RC1, and it works fine.

From the looks of it you are not using the latest version of configurator (10.7.0-RC1). Please downlowad and install the latest version from https://github.com/betaflight/betaflight-configurator/releases and try flashing again.

fuleinist, 2020-05-02:
Thanks @mikeller. just installed 4.2.0-RC1 as suggested but the problem still persists: FC led is solid red and gyro comes with no response after login.

```
Release info
Target: IFLIGHT_F722_TWING
Manufacturer ID: IFRC
Version: 4.2.0-RC1
Binary: betaflight_4.2.0_STM32F7X2-RC1.hex
Date: 30-04-2020 23:20
```

[image]

fuleinist, 2020-05-02:
Here is the working firmware ver I am using:

```
Target: EXF722DUAL 
Version: 4.1.1 
Binary: betaflight_4.1.1_EXF722DUAL.hex 
Date: 16-11-2019 00:07 
```

mikeller, 2020-05-02:
@fuleinist: Try clicking on 'Reset to defaults'.

fuleinist, 2020-05-02:
yep, that sorted it, Thanks a lot for your help!

closed 2020-06-08 by stale[bot]

DanielGlos, 2020-06-24:
Hi,
I'm having an issue with my FC, also iFlight SucceX-D F7 V2.1. There is solid red light on my FC when powered from battery or from USB. I can connect to betaflight, reciever is working I can arm and spin motors. DJI air unit is also working fine. Everything seems fine but the red light is still on. Any idea what this could be?

fuleinist, 2020-06-24:
What's in cli when you type in status?

DanielGlos, 2020-06-26:
@fuleinist 
This is the status output

MCU F722 Clock=216MHz, Vref=3.28V, Core temp=40degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3962, Max available config: 16384
Gyros detected: gyro 1, gyro 2
GYRO=ICM20689, ACC=ICM20689, BARO=BMP280
OSD: MAX7456
System Uptime: 54 seconds, Current Time: 2020-06-26T07:57:42.390+00:00
CPU:3%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 10
Voltage: 5 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 8
SD card: Not configured
Arming disable flags: RXLOSS CLI MSP

It says I2C errors 8 but I don't know what that means

bohdey, 2021-01-10:
Hi I have a succex-d mini and when I connect to the computer I get a solid green light and the FC doesn’t connect to beta flight!
I have tried different cables to connect some don’t do anything and others do the same light a solid green light and don’t connect to beta flight!
Tried plugging a lipo in aswell but no luck. 
My other quad with a succex f4 connects straight away no problems so the drivers are working

any ideas?

mikeller, 2021-01-10:
@bohdey: Sounds like a DOA - get a warranty replacement.