betaflight/betaflight #9754: Incorrect failsave BF 4.1.x with Flysky X6b receiver

DroiDnX, 2020-05-02:
The throttle value will not fall back to 885, will stay at 1000 when transmitter is turned of.

Only happens with Flysky X6b receiver. By going back to FW 4.0.6 the failsave works correct.

I tryed this with Foxeer f405 AIO and Matek f405 CTR. I tryed with three diffrent X6b receivers, all same issue.
With Fli14 or Flit10 receiver (same setup and configuration) the throttle value will go back to 885.

DroiDnX, 2020-05-02:
Added the diff.

etracer65, 2020-05-02:
It's doing exactly what you've configured it to do on failsafe.
```
rxfail 4 s 1000
```

DroiDnX, 2020-05-03:
No, this is the fallback value i have set for AUX1 / armswitch.
Fallback value for throttle is "Auto".
Also if i change failsave AUX 1 to 875 or hold the value for throttle will go down to 1000.
I flashed again, and didn't change anything, just arming.
Diff attached, same result.
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.6 Apr 25 2020 / 05:11:18 (283bda8bf) MSP API: 1.42
# manufacturer_id: FOXE   board_name: FOXEERF405   custom defaults: YES

# start the command batch
batch start

board_name FOXEERF405
manufacturer_id FOXE

# serial
serial 1 64 115200 57600 0 115200

# aux
aux 0 0 0 1650 2050 0 0

# master
set acc_calibration = -11,23,205
set serialrx_provider = IBUS

profile 0

rateprofile 0

# end the command batch
batch end

mikeller, 2020-05-03:
@DroiDnX: You need to set up failsafe correctly on your TX, see https://github.com/betaflight/betaflight/issues/8877#issuecomment-568616653 for an explanation.

Duplicate of #8877.

closed 2020-05-03 by mikeller

DroiDnX, 2020-05-03:
Thank you for your great support! This solved my problem.