betaflight/betaflight #9755: OSD element "VTX channel" delayed updating power level

carloscampos775, 2020-05-02:
OSD Element "VTX channel" delayed updating power level when using Low Power Disarm. Power level will switch in the OSD 45 seconds to 90 seconds after arming. However, the VTX is switching power levels when armed just not updating the OSD

```
# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr 25 2020 / 05:17:45 (212dab7ec) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF4V2, version: e15d1efa, date: 2019-12-30T03:57:28Z

# start the command batch
batch start

board_name KAKUTEF4V2
manufacturer_id HBRO

# name: CarlosFPV

# feature
feature GPS

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 3 2048 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# aux
aux 0 0 2 1775 2100 0 0
aux 1 46 3 1900 2100 0 0
aux 2 13 0 1400 1600 0 0
aux 3 35 0 1850 2100 0 0
aux 4 49 1 1850 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 30
vtxtable powerlabels 25 100 400 1W

# master
set gyro_lowpass2_hz = 300
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 240
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = 18,19,99,1
set mag_hardware = NONE
set min_check = 1005
set serialrx_provider = CRSF
set dshot_idle_value = 650
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set vbat_scale = 111
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_set_home_point_once = ON
set gps_rescue_initial_alt = 25
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1500
set pid_process_denom = 2
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_vbat_pos = 2401
set osd_rssi_pos = 26
set osd_link_quality_pos = 2134
set osd_tim_2_pos = 2454
set osd_vtx_channel_pos = 2123
set osd_craft_name_pos = 2113
set osd_gps_speed_pos = 2209
set osd_gps_sats_pos = 2164
set osd_home_dir_pos = 2132
set osd_home_dist_pos = 2145
set osd_altitude_pos = 2177
set osd_avg_cell_voltage_pos = 2433
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5880
set vcd_video_system = NTSC
set gyro_1_align_yaw = 2700
set name = CarlosFPV

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set anti_gravity_gain = 5000
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set p_pitch = 45
set i_pitch = 99
set d_pitch = 37
set f_pitch = 125
set i_roll = 94
set f_roll = 119
set i_yaw = 99
set f_yaw = 119
set idle_min_rpm = 14

rateprofile 0

# rateprofile 0
set roll_rc_rate = 152
set pitch_rc_rate = 152
set yaw_rc_rate = 152
set roll_expo = 47
set pitch_expo = 47
set yaw_expo = 14
set roll_srate = 57
set pitch_srate = 57
set yaw_srate = 57

# end the command batch
batch end
```
```
# resource
resource BEEPER 1 C09
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource PPM 1 C07
resource LED_STRIP 1 C08
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource INVERTER 3 B15
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 B05
resource LED 2 B04
resource LED 3 B06
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 C07
resource ADC_BATT 1 C03
resource ADC_RSSI 1 C01
resource ADC_CURR 1 C02
resource FLASH_CS 1 B03
resource OSD_CS 1 B14
resource GYRO_EXTI 1 C05
resource GYRO_CS 1 C04
resource USB_DETECT 1 A08
```

**Setup / Versions**
Kakute F4 (V2)
TBS Unify Pro32 with Crossfire Nano RX

[image]
[image]

mikeller, 2020-05-02:
@carloscampos775: So to clarify you use `vtx_low_power_disarm` to reduce VTX power when disarmed, and increase it when arming, and the power level switching itself works, but the display of the power level is lagging after arming?

This could be a side effect of VTX update messages being suppressed when arming to avoid interference with the audio channel of the VTX.

carloscampos775, 2020-05-03:
That is correct. Only the display of the power level is lagging after
arming.

codecae, 2020-05-06:
This could possibly also be some kind of logic voltage/impedence issue due to the shared audio line.  If the actual audio feed is of no importance, maybe try a 10k pull-down resistor from the smartaudio line to ground and see if it has a desired effect.

closed 2020-06-12 by stale[bot]