betaflight/betaflight #9760: Separate Motor Idles for Multiple Flight Modes/Voltages

That-One-Tiny-Whoop-guy, 2020-05-03:
I am trying to set up a brushless Tiny Whoop to have no motor idle on horizon mode (for indoor flying) but have some motor idle on acro mode (for freestyle). I think it would be a good idea to have the ability in Betaflight to set different motor idles for different flight modes and/or battery voltages and/or AUX channel values so you can fly outside of a window in horizon mode, switch to acro, and instantly begin performing acrobatics.

mikeller, 2020-05-04:
@That-One-Tiny-Whoop-guy: Not quite sure what you mean by this - why would you not set motor idle to the lowest speed that makes your motors spin reliably for any type of flying?

ggmaniack, 2020-05-04:
Well, disregarding the modes, the voltages do make sense, as throttle is relative to input voltage. It would be nice to be able to the old style IDLE (not dynamic) value per PID profile.

That-One-Tiny-Whoop-guy, 2020-05-04:
> @That-One-Tiny-Whoop-guy: Not quite sure what you mean by this - why would you not set motor idle to the lowest speed that makes your motors spin reliably for any type of flying?

That's what I'm currently doing, but for indoor flying, it's nice to be able to cut the motors off with just the throttle stick in case you get wrapped around something.

mikeller, 2020-05-06:
@That-One-Tiny-Whoop-guy: Ah what you actually want is for motors to stop when throttle is all the way down?

That-One-Tiny-Whoop-guy, 2020-05-06:
> @That-One-Tiny-Whoop-guy: Ah what you actually want is for motors to stop when throttle is all the way down?

Not exactly. In Acro mode I want 0 throttle to still spin the motors, but in Horizon mode I want 0 throttle to actually be 0 throttle.

mikeller, 2020-05-07:
@That-One-Tiny-Whoop-guy: Can you define what you mean by '0 throttle'?

That-One-Tiny-Whoop-guy, 2020-05-07:
> @That-One-Tiny-Whoop-guy: Can you define what you mean by '0 throttle'?

By "0 throttle" I mean the lowest stick position. In certain modes I want 0 throttle to still send power to the motors, creating an idle, while in other modes I want 0 throttle to send no power to the motors.

mikeller, 2020-05-09:
@That-One-Tiny-Whoop-guy: This sounds exactly what I proposed in https://github.com/betaflight/betaflight/issues/9760#issuecomment-624651269.

You can already do this (or something useful for your purpose):
Set up your idle to spin the motors like they should, then enable motor stop (`feature motor_stop`), disable permanent air mode (`feature -airmode`). After this you will have a mode switch available labelled 'AIR MODE'. Enabling this will mode will spin the motors, disabling it will stop them.

squshy7, 2020-05-12:
@mikeller to bring it back around to the voltage idle discussion, does dshot_idle_value respect motor_output_limit? because otherwise users that run 2 different profiles based on cell count will have either too high or too low of an idle for a certain cell count.

ctzsnooze, 2020-05-19:
DShot Idle is a percentage, so the actual idle RPM varies with battery cell count anyway.  Typically motor output limit would result in lower effective voltage from the higher cell count, bringing the idle RPM back closer to that of the lower cell count battery.  
The bigger issue is whether DShot idle value should be stored in a profile along with the dynamic idle value.

closed 2020-06-27 by stale[bot]