betaflight/betaflight #9761: Aikon F7 Mini doesn't work

CapVille, 2020-05-03:
After flashing to 4.2 with the 10.7 configurator the FC doesn't receive configuration, after setting ports I press save and reboot and nothing happens, I try to change to configuration tab and it stays loading.

etracer65, 2020-05-03:
Flash the flight controller again and be sure to select "full chip erase". Make sure that when you connect after flashing you answer "yes" to applying custom defaults.

CapVille, 2020-05-03:
I did it that way, I used full chip erase. couple of times an didn't work. the FC works fine with 4.1.5

mikeller, 2020-05-04:
@CapVille: Just tested with an Aikon F7 mini, and 4.2-RC1 installs and works just fine with 10.7.0-RC1.

> the FC doesn't receive configuration, after setting ports I press save and reboot and nothing happens

Can you please describe exactly the steps that you are taking, and post a screenshot of the configurator window right before you hit 'Save & Reboot'?

CapVille, 2020-05-04:
After flashing the FC, I start the process to configure it from 0, I start in the port tab and after pressing save and reboot nothing happens, but I try to change to Configuration tab and it stays loading.
[image]

[image]

mikeller, 2020-05-06:
Tested exactly this configuration, and it's working just fine for me:
[image]
Can you try uninstalling configurator and re-installing it? But we are really clutching at straws now...

CapVille, 2020-05-06:
I am using MacOS, does it affect? I will try to re-install it, and I also will try in a Windows

mikeller, 2020-05-06:
@CapVille: Could be that it's MacOS specific - I don't have a way to test this, so feedback is appreciated.

CapVille, 2020-05-06:
I tried to re-install it and didn't work, but I tried it in Windows and it worked perfect.

studioratte, 2020-05-06:
I had the same problem after flashing the aikon f7 mini v2 (DJI Version) to betaflight 4.2 on configurator 10.7.0 running on Mac OSX 10.12.6. (Tried twice with full flash erase)
Unfortunately i got no Windows system to try.
Was able to flash down to BF 4.1.6 and will be flying 4.1 until issue is solved.

On my other quad the hobbywing F4G3 was updated easy and works like a charm. (same Mac OS Computer and betaflight configurator version)

etracer65, 2020-05-07:
I've been able to replicate this and I'm looking into it. So far it only seems to be affecting AikonF7 flight controllers with MacOS. At this point it's a connection that doesn't make much sense.

Dirtbrother, 2020-05-09:
I'm having the exact same issue on a TMotor F7 with OSx.

Absolutely fine with 4.1.6 but soon as i go to 4.2 it freezes on save & reboot.

Did full chip erase and applied defaults and no change. Flashed back to 4.1.6 and all good again.

etracer65, 2020-05-09:
I have a fix for this problem that I'd like people to test. Please flash this firmware and test that the Configurator now behaves normally. The firmware file is only appropriate for F722 based flight controllers (like the AikonF7).

[betaflight_4.2.0_STM32F7X2_af17ef2.hex.zip](https://github.com/betaflight/betaflight/files/4604865/betaflight_4.2.0_STM32F7X2_af17ef2.hex.zip)

If you don't know how to flash a development version of a unified target, please watch this video:

https://youtu.be/I1uN9CN30gw

CapVille, 2020-05-09:
I tried it and at least it allows to configure the quad, no more freeze on save and reboot. I have to try it flying 
Thanks

etracer65, 2020-05-09:
@CapVille Thanks. Flight testing is not required. There are only changes to the communication with the Configurator.

CapVille, 2020-05-09:
Perfect, thank you.

Dirtbrother, 2020-05-10:
@etracer65 just flashed your fix file on my TMotor F7 and can confirm it no longer has the issue.

Will go back to 4.1.6 for now until new RC is out.

Thanks for sorting this man!

closed 2020-05-13 by mikeller

GaelGuegan, 2020-07-04:
@etracer65  I don't know if it is linked with this issue, but I cannot change the setting "beeper_frequency" on my mamba mini f722. I am using the last betaflight release 4.2.0 and with the CLI I do a "save" and after reconnecting my setting didn't change. (all other settings are correctly saved)

PR #9789 (2020-05-13): Fix HAL USB vcp serial 64 byte packet receive bug
Fixes #9761 
Fixes https://github.com/betaflight/betaflight-configurator/issues/2006

Fixes a bug in the HAL USP vcp serial logic that would cause communication to hang after receiving a packet containing exactly 64 bytes.

When a packet is exactly 64 bytes the USB specification requires an empty (0 byte) pack to be additionally sent. The receiving logic was not handling this 0 byte packet properly. The logic error would cause it to not trigger receiving the next packet and communication would stop.